itemisCREATE/statecharts #1019: enable Automated Error Reporting (AERI) for yakindu plugins

yuecelm, 2016-10-07:
https://wiki.eclipse.org/EPP/Logging

andreasmuelder, 2016-10-07:
We tried to add the automated error reporting last year. At that point it was not possible to add custom servers / endpoints except the eclipse bugzilla. Thanks for the link, it seems that it is now possible for us to integrate this nice feature!

BeckmaR, 2016-10-18:
The next step will be to discuss the platform on which we want to collect the reported Bugs. We can basically implement any platform we want, because AERI allows us to have our own settings and error handler.
While GitHub would be the first solution coming to mind, it has downsides:
An authenticated user is needed to create issues. Either the SCT user has to login himself and generate a login token for us, or we would need to have a bot-user that logs in and posts all issues anonymously. The downside of the first solution is that not all users will have a GitHub-account or feel like logging in, even with the many possibilities today like application rights etc many people feel spied on. The other solution forces us to put the login token into the source code, which is as good as a username and a password - we absolutely should not do that.

Another option would be to have an own BugZilla instance on our servers where bugs are then collected, but this and any other options mean we need to manually go through them and create issues in GitHub. The sheer amount of issues that could come in might be a good reason not to use this GitHub repo at all but a separate one, so this point won't go away.

Another options is a SaaS named CtrlFlow. It looks like these guys were behind the implementation of AERI. They offer a service, server included, that can handle the error reports, implementing a Bugtracker with sync possibilities, including GitHub. That might actually be nice, but the price goes from free to over 50€/month to 250€/month [really quick](https://www.ctrlflow.com/automated-error-reporting/plans-and-prices/).

Another option is email. We can make SCT open the user's email client when he agrees to send an error report. Depending on the size of our user base (and the amount of bugs), that might be a valid option.

For numbers: the whole eclipse project received up to 80.000 error reports per week at times.

Options summarized:
* GitHub - involves messy authentication process and a huge load of issues in some repo. If a separate repo is set up solely to collect bugs, a copy process per hand is needed when bugs are valid.
* BugZilla / similar - authentication process unclear. If there's none, someone will find a way to post stuff there we don't want. If there is an authentication process, similar issues like with GitHub will arise. Separate software needed, more segmentation in the bug tracking process. Also, BugZilla doesn't look like something @andreasmuelder might love.
* CtrlFlow - the process seems to be nice, the interface looks simple, no server needed, painless, but not for free. The ease of use might be worth it.
* email. Similar issues regarding copy-paste of bugs, but allows us to get back to users reporting errors. Might be messy and unorganized when more than ~10 bugs per week come in.

andreasmuelder, 2016-11-10:
After some testing the solution provided by ctrlflow seems to be the most promising. 

AERI will be part of the next release. The implementation details should be discussed in our internal issue tracker. Closing this.

closed 2016-11-10 by andreasmuelder

yuecelm, 2016-11-10:
how we handle error reports for custom plugins? in the next release it will be sended generally to your ctrlflow account. can we get some access there when some errors with our custom plugin (matches package name com.siemens.*) are triggered?

andreasmuelder, 2016-11-10:
@yuecelm professional support is only available for our customers, I recommend to get in touch with our sales manager panagiotidis@itemis.com to discuss this topic!