itemisCREATE/statecharts #1030: Saving big model file takes too long

tkutz, 2016-10-12:
Take the model from manual-tests/org.yakindu.sct.test.manual/testmodels/big_file.sct

Change something and save it.

The "Building workspace" job starts and hangs at "invoking build participants" for a long time. This is way too long.

tkutz, 2016-10-13:
Most time seems to be spent for painting the figures again.

[image]

andreasmuelder, 2016-11-07:
@tkutz Can you still reproduce this with the 2.8 release? At least on my windows machine it is quite fast.

andreasmuelder, 2017-05-08:
@tkutz is this still an issue?

BeckmaR, 2017-06-21:
@andreasmuelder Yakindu/sctpro#773 says yes, it is

andreasmuelder, 2017-08-31:
Closing this, fixed in Pro

closed 2017-08-31 by andreasmuelder