itemisCREATE/statecharts #1046: Custom generator does not compile any more

stmork, 2016-10-18:
It looks like API changes:
```
[ERROR] 
ERROR:  QtGenerator.xtend - 
15: org.yakindu.sct.generator.core.AbstractWorkspaceGenerator cannot be resolved to a type.
[ERROR] 
ERROR:  QtGenerator.xtend - 
17: org.yakindu.sct.generator.core.impl.IExecutionFlowGenerator cannot be resolved to a type.
[ERROR] 
ERROR:  QtGenerator.xtend - 
26: AbstractWorkspaceGenerator cannot be resolved to a type.
[ERROR] 
ERROR:  QtGenerator.xtend - 
26: IExecutionFlowGenerator cannot be resolved to a type.
[ERROR] 
ERROR:  QtGenerator.xtend - 
33: The method generate(ExecutionFlow, GeneratorEntry, IFileSystemAccess) of type QtGenerator must override a superclass method.
[ERROR] 
ERROR:  QtGenerator.xtend - 
26: Superclass must be a class
[ERROR] 
ERROR:  QtGenerator.xtend - 
26: Implemented interface must be an interface
```

tkutz, 2016-10-18:
Sorry for that, but for version 2.8.0 we have done some code refactorings in our generator infrastructure, among which we also changed the API for custom generators. 

You need to implement IExecutionFlowGenerator like the following (note the different package):
````
public class MyCustomGenerator implements org.yakindu.sct.generator.core.IExecutionFlowGenerator {
	public void generate(ExecutionFlow flow, GeneratorEntry entry, IFileSystemAccess fsa) {
            // add here your generator logic
	}
}
````

stmork, 2016-10-18:
Yes, I know. But I have to implement _GenericJavaBasedGenerator_ and _AbstractWorkspaceGenerator_ to get the standalone build running. I assume, that your standalone build example is also not compiling any more.

tkutz, 2016-10-18:
No, since we changed the generator infrastructure, our old standalone example is not working anymore. We are working on a proper solution here and will publish more information (blogs, documentation) later. 

What exactly do you need from `GenericJavaBasedGenerator` and `AbstractWorkspaceGenerator`? These were classes that actually handled the execution of generators not the generation itself. Some of their logic is now reflected in `org.yakindu.sct.generator.runner.WorkspaceGeneratorExecutor`. But normally, you wouldn't need to take care about it.

You can see an example of a custom generator (although in this case based on SGraph model instead of ExecutionFlow) in this [blog post](https://blogs.itemis.com/en/custom-code-generator-scxml). There is also an example (SCXML) which you can directly import in your IDE.

stmork, 2016-10-18:
I got the standalone builder running with those classes. If there is an alternative or a best practice it's OK for me. For the first I disabled my standalone projects and I'm excited to read about the standalone update post.

closed 2016-11-07 by andreasmuelder