itemisCREATE/statecharts #1050: Custom generator does not run any more

stmork, 2016-10-18:
After invoking the custom generator a dialog with a NullPointerException appears with no stack trace. The console show the following stack trace:
```
org.eclipse.core.runtime.CoreException: Executable extension definition for "bindings" not found.
        at org.eclipse.core.internal.registry.ConfigurationElement.throwException(ConfigurationElement.java:62)
        at org.eclipse.core.internal.registry.ConfigurationElement.createExecutableExtension(ConfigurationElement.java:222)
        at org.eclipse.core.internal.registry.ConfigurationElementHandle.createExecutableExtension(ConfigurationElementHandle.java:55)
        at org.yakindu.sct.generator.core.extensions.GeneratorExtensions$GeneratorDescriptor.getBindings(GeneratorExtensions.java:74)
        at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.createInjector(GeneratorExecutorLookup.java:76)
        at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.createExecutor(GeneratorExecutorLookup.java:62)
        at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.execute(GeneratorExecutorLookup.java:50)
        at org.yakindu.sct.generator.builder.EclipseContextGeneratorExecutorLookup$1.run(EclipseContextGeneratorExecutorLookup.java:43)
        at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
```

Even a new custom generator inside a fresh workspace setup using your documentation shows the same behavior.

tkutz, 2016-10-18:
Am I right in the assumption that you used the "Configure for Plugin Export" option?
This is indeed not properly working right now and will be fixed soon.

We changed the extension point schema for registering a custom generator. Now it is required to provide a GeneratorModule under the "bindings" key. The plugin.xml that is generated for a custom generator project is indeed incomplete in this regard.

For a workaround you can do the following:
* In plugin.xml, add a **bindings** class like here:
````
   <extension
         point="org.yakindu.sct.generator.core.generator">
      <SCTGenerator 
	    executor="org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor"
            <!-- Module containing generator bindings -->
            bindings="custom.generator.CustomGeneratorModule" 

            description="A custom code generator"
            id="custom::sctGenerator"
            name="CustomGenerator"
            contentType="statechart"
            elementRefType="org.yakindu.sct.model.sgraph.Statechart">
            <FeatureLibrary
            	library_id="org.yakindu.generator.core.features">
            </FeatureLibrary>
      </SCTGenerator>
   </extension>
````
* Implement CustomGeneratorModule in package custom.generator as follows:
````
package custom.generator;

import org.yakindu.sct.generator.core.GeneratorModule;
import org.yakindu.sct.generator.core.IExecutionFlowGenerator;
import org.yakindu.sct.model.sgen.GeneratorEntry;

import com.google.inject.Binder;

public class CustomGeneratorModule implements GeneratorModule {

	public void configure(GeneratorEntry entry, Binder binder) {
		binder.bind(IExecutionFlowGenerator.class).to(CustomGenerator.class);
	}
}
````

This should solve the NPE and make the generator run again!

tkutz, 2016-10-26:
Fixed with pull request #1057

closed 2016-10-26 by tkutz

tkutz, 2016-11-04:
We have just released an intermediate version 2.8.1, where this issue is fixed.

stmork, 2016-11-05:
OK. I've got the custom generator running again. Thank you.