itemisCREATE/statecharts #1060: error dialog by save

DominikTesch, 2016-10-26:
[image]

[default2.zip](https://github.com/Yakindu/statecharts/files/553006/default2.zip)

I get this error message after saveing the project.
You have a few simple subdiagrams in each other.

DominikTesch, 2016-10-27:
[image]
[image]

the first error you will get when you have a lot of subrigions ( more than 12 )

the second one i can't reproduce a second time

andreasmuelder, 2016-11-07:
@DominikTesch I can't reproduce this, please add some steps to reproduce.

DominikTesch, 2016-11-10:
java.lang.NullPointerException
	at org.eclipse.emf.transaction.util.TransactionUtil.getEditingDomain(TransactionUtil.java:81)
	at org.eclipse.gmf.runtime.emf.core.util.EMFCoreUtil.getImports(EMFCoreUtil.java:244)
	at org.yakindu.sct.ui.editor.partitioning.ResourceUnloadingTool.unloadEditorInput(ResourceUnloadingTool.java:55)
	at org.yakindu.sct.ui.editor.partitioning.DiagramPartitioningDocumentProvider.disposeElementInfo(DiagramPartitioningDocumentProvider.java:128)
	at org.eclipse.gmf.runtime.diagram.ui.resources.editor.document.AbstractDocumentProvider.disconnect(AbstractDocumentProvider.java:425)
	at org.eclipse.gmf.runtime.diagram.ui.resources.editor.parts.DiagramDocumentEditor.disposeDocumentProvider(DiagramDocumentEditor.java:519)
	at org.eclipse.gmf.runtime.diagram.ui.resources.editor.parts.DiagramDocumentEditor.dispose(DiagramDocumentEditor.java:217)
	at org.yakindu.sct.ui.editor.partitioning.DiagramPartitioningEditor.dispose(DiagramPartitioningEditor.java:173)
	at org.yakindu.sct.ui.editor.editor.StatechartDiagramEditor.dispose(StatechartDiagramEditor.java:302)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.invalidate(CompatibilityPart.java:237)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.destroy(CompatibilityPart.java:403)
	at sun.reflect.GeneratedMethodAccessor130.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:966)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:931)
	at org.eclipse.e4.core.internal.di.InjectorImpl.uninject(InjectorImpl.java:212)
	at org.eclipse.e4.core.internal.di.Requestor.uninject(Requestor.java:170)
	at org.eclipse.e4.core.internal.contexts.ContextObjectSupplier$ContextInjectionListener.update(ContextObjectSupplier.java:84)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.update(TrackableComputationExt.java:111)
	at org.eclipse.e4.core.internal.contexts.EclipseContext.removeListenersTo(EclipseContext.java:470)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.uninject(ContextInjectionFactory.java:144)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeRemoveGui(PartRenderingEngine.java:927)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$1(PartRenderingEngine.java:846)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$3.run(PartRenderingEngine.java:841)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.removeGui(PartRenderingEngine.java:825)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.subscribeTopicToBeRendered(PartRenderingEngine.java:173)
	at sun.reflect.GeneratedMethodAccessor126.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	at org.eclipse.e4.core.di.internal.extensions.EventObjectSupplier$DIEventHandler.handleEvent(EventObjectSupplier.java:80)
	at org.eclipse.equinox.internal.event.EventHandlerWrapper.handleEvent(EventHandlerWrapper.java:201)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:197)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:148)
	at org.eclipse.equinox.internal.event.EventAdminImpl.dispatchEvent(EventAdminImpl.java:135)
	at org.eclipse.equinox.internal.event.EventAdminImpl.sendEvent(EventAdminImpl.java:78)
	at org.eclipse.equinox.internal.event.EventComponent.sendEvent(EventComponent.java:39)
	at org.eclipse.e4.ui.services.internal.events.EventBroker.send(EventBroker.java:94)
	at org.eclipse.e4.ui.internal.workbench.UIEventPublisher.notifyChanged(UIEventPublisher.java:60)
	at org.eclipse.emf.common.notify.impl.BasicNotifierImpl.eNotify(BasicNotifierImpl.java:374)
	at org.eclipse.e4.ui.model.application.ui.impl.UIElementImpl.setToBeRendered(UIElementImpl.java:303)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.hidePart(PartServiceImpl.java:1350)
	at org.eclipse.ui.internal.WorkbenchPage.hidePart(WorkbenchPage.java:1590)
	at org.eclipse.ui.internal.WorkbenchPage.closeEditors(WorkbenchPage.java:1503)
	at org.eclipse.ui.internal.WorkbenchPage.closeEditor(WorkbenchPage.java:1636)
	at org.yakindu.sct.ui.editor.partitioning.DiagramPartitioningUtil.closeSubdiagramEditors(DiagramPartitioningUtil.java:226)
	at org.yakindu.sct.ui.editor.edithelper.RegionEditHelper$1.doExecute(RegionEditHelper.java:39)
	at org.eclipse.emf.workspace.AbstractEMFOperation.execute(AbstractEMFOperation.java:150)
	at org.eclipse.emf.workspace.CompositeEMFOperation.doExecute(CompositeEMFOperation.java:217)
	at org.eclipse.emf.workspace.AbstractEMFOperation.execute(AbstractEMFOperation.java:150)
	at org.eclipse.emf.workspace.CompositeEMFOperation.doExecute(CompositeEMFOperation.java:217)
	at org.eclipse.emf.workspace.AbstractEMFOperation.execute(AbstractEMFOperation.java:150)
[... cut ...]
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1492)

DominikTesch, 2016-11-10:
[default.zip](https://github.com/Yakindu/statecharts/files/583147/default.zip)

open the statechart
select StateA, Transition and StateB ( one state must have a subdiagram )
Press Ctrl and C
Select the region 
Press Ctrl and V
Undo with Ctrl and Z
Press Ctrl and V again
Save with Ctrl and S
Now you get the first error massage

https://cloud.githubusercontent.com/assets/15800400/19722928/2f8f84ba-9b79-11e6-8f94-1e2cb2eff863.png

After reload the programm you get this error massage
[image]

andreasmuelder, 2017-01-23:
@DominikTesch Can you please check if this also happens with the bundles product or maybe this is a side effect of Papyrus.

RainerKlute, 2017-03-17:
I could reproduce with 2.9.3 what @DominikTesch [in this comment](https://github.com/Yakindu/statecharts/issues/1060#issuecomment-259659932).

I was **not** able to reproduce that behavior with a runtime of the current master. However, [Ctrl+V] doesn't work properly in the current master! I could not copy/paste the [State A, transition, State B] combo nor State A alone. However, I could copy/paste State B without problems.

So something is definitely broken here one way or the other.

terfloth, 2017-06-20:
This issue seems to be solved!
   * revalidate with master
   * derive copy paste issues into separate task

RainerKlute, 2017-06-21:
Well, unfortunately it isn't solved. When pressing ```[Ctrl]+S```, the following error message appears:

[image]

(I tried with the current YSCT PRO product.)

BeckmaR, 2019-02-26:
This is not related to the actual reverting of the pasting at all.

Before pasting, one has to select the region where the states are meant to be pasted into. When doing CTRL + Z, _the region is silently unselected_. When pressing CTRL + V again, with no region selected, the elements just go onto the canvas. This can be reproduced with all elements - select them and just paste them. This will flag the statechart as *dirty*, even though nothing actually happened. Only if one of the states that were pasted into nirvana is a subdiagram, this error occurs.

So, the actual steps to reproduce are these:

* Create a new state
* Add a subregion
* Refactor into subdiagram
* Copy the state / subdiagram
* Select the canvas (nothing) and paste - statechart gets flagged as dirty
* Save.

BeckmaR, 2019-02-26:
The newer issue #2264 found the same problem and explains it way more concise. Closing this in favor of the new one.

closed 2019-02-26 by BeckmaR