itemisCREATE/statecharts #1062: String handling in C generator needs to be reworked

BeckmaR, 2016-10-26:
The handling of strings in C is somewhat broken. A few problems include, but are not limited to:
* Transferring of strings into the statechart copies the pointer instead of allocating a new one. Changing the original string changes the "copied" one.
* String concatenating is not possible.
* Arrays are not supported in SCT OSE, but that's what strings in C are. Real handling of them is prohibited as such. Allocating buffers is not possible.
* Even if I `malloc` new strings, I can't free them because I can never know when SCT doesn't need them anymore.

Basically we should either drop strings completely in OSE or we should provide a real implementation with our own sc_string utility functions. Currently there is no way to operate with a SCT statechart with strings coming from the outside.

ThomasBbg, 2016-12-01:
The handling of strings in C++ has the same problems, as the same code is generated as for C.
In C++ the fix would be easier, as there is the native type std::string, that behaves similar to the Java String.

andreasmuelder, 2018-01-11:
Closing this as duplicate of #1233

closed 2018-01-11 by andreasmuelder