itemisCREATE/statecharts #1064: Is there any equivalent of 'else'?

robfox1976, 2016-11-02:
In a state, I want to get something like;

> entry [myNumber < 10]/
> TaskA()
> 
> **_else_**
> TaskB()

Is there any way of doing this, as I think it's messy to have;

> entry [myNumber < 10]/
> TaskA()
> 
> entry [myNumber >= 10]/
> TaskB()

But at the moment, this is the only way I see of doing this.

andreasmuelder, 2016-11-02:
@robfox1976 we have a conditional expression,
entry / myNumber < 20 ?   TaskA() : TaskB()

robfox1976, 2016-11-03:
Thanks for your reply. As I'm not running with the Pro version, I tried the following;

```
entry/
user.x>5 ? user.y = 1: user.y = 2
```
However, it really did not like this, it errors "mismatched input '=' expecting ':'"
Is my problem that I am not using Pro, or an older version, or that I can only call a function, rather than trying to make an assignment?

andreasmuelder, 2016-11-03:
you can assign a conditional expression to a variable:
entry/ user.y = user.x>5 ?  1 : 2
@RainerKlute We should improve our documentation on this topic and add some examples

robfox1976, 2016-11-03:
Perfect, that does exactly what I need it to do. Thanks very much.

closed 2016-11-07 by andreasmuelder