itemisCREATE/statecharts #1068: Raising internal events not reported after runCycle()

stmork, 2016-11-08:
When raising internal events there is no chance to evaluate any occurance of them after calling _runCycle()_. I use the following statemachine as an example:

[image]

And using following C++ code using the ```innerFunctionVisibility = "protected"``` property feature. The example tries to evaluate the so named _never_ event which is always false. I assume that the clearInEvents() method at the end of the runCycle() method accidently clears the internal raised flags.

```c++
#include <stdlib.h>
#include <stdio.h>
#include "src-gen/Bug01.h"

class StateMachine : public Bug01, public Bug01::InternalSCI_OCB
{
public:
	StateMachine()
	{
		setInternalSCI_OCB(this);
	}

	void start() override
	{
		printf("start()\n");
	}

	void trigger()
	{
		printf("trigger()\n");
		ifaceInternalSCI.raise_trigger();
		printf("runCycle()\n");
		runCycle();
		printf("never raised? %s\n",
			ifaceInternalSCI.isRaised_never() ? "true" : "false");
	}
};

int main(int argc, char *argv[])
{
	StateMachine sm;

	printf("-----------------\n");
	sm.init();
	sm.enter();
	sm.trigger();
	sm.exit();
	printf("-----------------\n");

	return EXIT_SUCCESS;
}
```
This code produces the output:
```
-----------------
start()
trigger()
runCycle()
never raised? false
-----------------
```
But I expected: ```never raised? true```.

BeckmaR, 2016-11-22:
Hi stmork,

the behaviour of internal events is well documented - they are only visible in regions and states processed after the raising of the event, which you can control by setting the region priorities.

Further, not being able to see internal events from the outside is intended behaviour. 
Try the following:

    interface:
      out event never

This way, you will be able to check the event with the isRaised-funtcion.

Regards,
Rene

closed 2016-11-22 by BeckmaR

stmork, 2016-11-22:
IMO loosing events whether internal or not is a misbehaviour. Even if documented like it works now. The raised flag should only be cleared if consumed inside the statemachine to give the surrounding application the chance to rerun the runCycle() method.