itemisCREATE/statecharts #1069: Example needed for CSharpGenerator

JackUkleja, 2016-11-09:
Hi,

I have been experimenting trying to get the org.yakindu.sct.generator.csharp generator working. I am a .NET developer so have no idea what I'm doing and need help :)

I have YAKINDU 2.8.0 and I have downloaded the source for org.yakindu.sct.generator.csharp and placed that as a project in my workspace, then I have created another project with an .sgen and a .sct file (using light_switch example). 

I have the .sgen setup as follows:

```
GeneratorModel for yakindu::generic {
	statechart Light_Switch {
		feature Outlet {
			targetProject = "test"
			targetFolder = "output"
		}
		feature Generator {
			generatorProject = "org.yakindu.sct.generator.csharp"
			generatorClass = "org.yakindu.sct.generator.csharp.CSharpGenerator"
		}
	}
}
```

But when I Generate Code Artifacts for this .sgen I get:

> Generating 'Light_Switch' to target project ...test
> java.lang.NullPointerException
> 	at org.yakindu.sct.generator.csharp.CSharpGenerator.generate(CSharpGenerator.java:84)
> 	at org.yakindu.sct.generator.runner.WorkspaceGeneratorExecutor.execute(WorkspaceGeneratorExecutor.java:51)
> 	at org.yakindu.sct.generator.core.execution.AbstractGeneratorEntryExecutor.execute(AbstractGeneratorEntryExecutor.java:59)
> 	at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.execute(GeneratorExecutorLookup.java:51)
> 	at org.yakindu.sct.generator.builder.EclipseContextGeneratorExecutorLookup$1.run(EclipseContextGeneratorExecutorLookup.java:43)
> 	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
> done...

I am guessing that maybe `yakindu::generic` is the wrong type for the .sgen so I guessed maybe it was `yakindu::csharp` or `yakindu::cs` but neither work.

Can anyone give advice or just check in a working example into the examples repo?

Thanks,
Jack

andreasmuelder, 2016-11-09:
@JackUkleja Note that the CSharp code generator was contributed by @m-sonntag some time ago and is not (yet) part of the official Statechart Tools release. We didn't have the time yet to implement the test suite. Use it at own risk and please report any bugs you may encounter.

The simplest way to get started is to download the latest statechart tools release 2.8.0. You can download the csharp code generator attached to this ticket. rename the file extension from .zip to .jar and copy it to yakindu-sct/plugins folder.

Then you can use the csharp generator like this:

GeneratorModel for yakindu::csharp {

	statechart default {

		feature Outlet {
			targetProject = "Test"
			targetFolder = "src-gen"
			libraryTargetFolder = "src"
		}
	}
}

[org.yakindu.sct.generator.csharp_2.8.1.201611091054.zip](https://github.com/Yakindu/statecharts/files/580461/org.yakindu.sct.generator.csharp_2.8.1.201611091054.zip)

JackUkleja, 2016-11-09:
Thanks @andreasmuelder. That seems to work - at least: it to generates C#.

Can you tell me what changes (if any) you needed to make to the C# generator source to make this work? 

Because I have tried to add the CSharpGenerator source as a project in my workspace, but that does not seem to get invoked properly. Must the generators be compiled to a JAR and put in the plugins dir for it to work? Ideally we would be able to edit and test changes to the C# generator without that step.

Happy to provide pull requests to the C# generator once we figure out how to get things up and running. I think there will be high demand for this in the .NET community.

andreasmuelder, 2016-11-10:
@JackUkleja great to hear that you want to contribute to the C# generator!

The best way to get started is to set up SCT from sources as described here (Build everything from source): https://github.com/Yakindu/statecharts/blob/master/README.md

Once your workspace compiles, you can start a new eclipse runtime from your developer workspace. (right click on any plugin and select Debug as / Eclipse Application). A new eclipse will start, you can test the changes to the csharp generator in this new runtime. This is the recommended way for plugin developers, the yakindu generic workspace executer does not work properly (yet) with the c# generator.

What's still missing for the C# Generator are unit tests. We developed a framework called SCTUnit that we use to write tests on the semantic level of statecharts to ensure that all code generators behave in the same way. The Testmodels and their corresponding sctunit tests can be found here:

https://github.com/Yakindu/statecharts/tree/master/test-plugins/org.yakindu.sct.test.models
\testmodels\SCTUnit -> our test statecharts
\test\ -> the sctunit tests

for Java, we generate JUnit tests from all sctunit files, see:
https://github.com/Yakindu/statecharts/tree/master/test-plugins/org.yakindu.sct.generator.java.test/test-gen/org/yakindu/sct/generator/java/test

The next step to get the C# code generator ready would be an SCTUnit code generator for C#Tests. 
Since we are not .NET guys - what is the .NET testing framework to choose? NUnit? Can those tests be executed on a linux build server? Maybe you could provide 1-2 example tests so we have a reference implementation and can implement a sctunit c# code generator?

JackUkleja, 2016-11-11:
We are having trouble working out how you generated the test cases for each specific language. Where are the test case generators for each supported language? Obviously as there are a lot of tests we don't want to be writing them by hand for C# either!

And how do you actually run "SCUnit"? Is this actually a command line tool, or is it just the name of the framework, test charts, and the generated test code (which are then just run as regular junit tests?).

Yes NUnit is a safe best for a testing framework. It's likely that there is nothing to prevent the generated C# code being ".NET Core" compatible, which means we can use the x-plat .NET Core framework to run all the tests, including on your CI system (which sounds like it's Linux?).

andreasmuelder, 2016-11-11:
@JackUkleja SCTUnit is a pro feature that will be released as part of our professional edition soon. For now, @BeckmaR started implementing a CSharp SCTUnit Test generator and we will provide the generated tests in the open source repository soon. We highly appreciate your feedback then!

BeckmaR, 2016-11-14:
@JackUkleja You can checkout the branch iss_1069, which contains an initial version of the generated tests for our test models. I had to adapt the csharp generator a tiny bit to make it work for the tests, you will find that version in the mentioned branch as well.
Attached you will find a zip file which you can import into your SCT runtime to try out the generated sources and tests. If you have any further questions, ideas or proposals, please do not hesitate to give us your feedback.
[csharp.zip](https://github.com/Yakindu/statecharts/files/588830/csharp.zip)
Some statemachines don't work properly yet, I just deleted them from my project in Monodevelop and threw away the relevant tests as well. All other tests were completed successfully.

andreasmuelder, 2016-11-24:
Closing this as an example was provided. Pull Requests welcome 👍

closed 2016-11-24 by andreasmuelder