itemisCREATE/statecharts #1076: Feature request: Draw expressions vertically on vertically orientated transitions

stmork, 2016-11-10:
If an expression is drawn vertically for vertical transitions a state chart gets more readable.

BeckmaR, 2016-12-09:
I can only speak for myself, but I do not think this is more readable. Maybe even less so.

[image]

stmork, 2016-12-10:
In your case there is enough place to store every text horizontally. I agree that horizontal positioning is more readable anyway. But I have a statechart with lots of vertical parallel transitions. So its difficult to distinguish which action belong to which transition.

Here is an idea of what I mean. The image is ententionally blurred.
[image]

rvilarl, 2017-04-01:
I would suggest to allow any orientation, not just horizontal and vertical. Having the possibility to change the orientation of the expression, in order to match the orientation of the arrow, would facilitate the visual association of both element.

svenjawendler, 2017-06-26:
I would suggest to leave the orientation as it is. But you can add linebreaks on transition with STRG + Return.