itemisCREATE/statecharts #1082: Bug: Deleting internal operations in the interface description removes necessary includes for tracing callback functions

lostwald, 2016-11-14:
Hi,

after removing an internal operation from the interface of a statechart model:

[image]

the necessary include of the statechart header is removed from the <statechart>_required.h. However, this Include is still needed for the state_entered & state_exited tracing callback functions. See diff:

[image]

The resulting code cannot be compiled.

BeckmaR, 2016-11-14:
Hey @lostwald,

thank you for the feedback. I could reproduce the issue; activating the tracing feature in the `*.sgen` file produces the `*Required.h`-file, but the main statechart header is not included there. We will fix this.

However, a workaround would be to include `OSM.h` and `OSMRequired.h` in the compilation unit where the tracing functions are defined. I have attached an example project for you that builds and runs on my linux machine. Please be advised that the toolchain will likely need to be reconfigured when you want to try it out on a windows machine.
[tracing.zip](https://github.com/Yakindu/statecharts/files/589163/tracing.zip)

Regards,
Rene

andreasmuelder, 2016-11-14:
Scheduled for Milestone 34

PR #1094 (2016-11-22): Add StateMachine Header when tracing is active
Fixes #1082
commit f5de3ad2 (2016-11-22): Add StateMachine Header when tracing is active (#1094)

Due to a request from GitHub-user Lostwald, the main Statemachine header is now
included when tracingEntryState or tracingExitState are set to true in the
*.sgen-file.
Fixes #1082
closed 2016-11-22 by andreasmuelder