itemisCREATE/statecharts #1109: Bug in raiseTimeEvent()

Anzey, 2016-11-26:
C++ generator created function 

void Glad::raiseTimeEvent(sc_eventid evid)
{
	if ((evid >= &timeEvents) && (evid < &timeEvents + sizeof(timeEvents)))
	{
		*(sc_boolean*)evid = true;
	}
}

Part:
**&**_timeEvents + sizeof(timeEvents)_
is wrong.
Should be (without ampersand):
_timeEvents + sizeof(timeEvents)_

stmork, 2016-11-27:
This issue fits perfectly into my still open issue #608 from April. The pointer arithmetics is not only insane but also it is not really object orientated. So:
+1

BeckmaR, 2016-11-28:
Hey @Anzey, I've allowed myself to edit your comment for better readability.

terfloth, 2016-12-07:
@Anzey you are right - will fix that soon.

terfloth, 2016-12-08:
The current fix will look like this:

```
void Test::raiseTimeEvent(sc_eventid evil)
{
	if ((evid >= (sc_eventid)timeEvents) && (evid < (sc_eventid)(&timeEvents[timeEventsCount])))
	{
		*(sc_boolean*)evid = true;
	}				
}
```

while `timeEventsCount` is defined as a constant and `sc_eventid` is redefined as:

`typedef intptr_t sc_eventid;`

closed 2016-12-11 by terfloth