itemisCREATE/statecharts #1113: Headless build does not work for community edition

stmork, 2016-11-28:
The headless build tool scc shows following response upon startup:
```
libra(neon/eclipse-jee) 14> ./scc -h
./scc: line 7: /opt/neon/eclipse-jee/SCT: No such file or directory
```

andreasmuelder, 2016-11-28:
@stmork I assume you did not download our community edition product bundle but installed SCT via the update site instead. Then, you do not have an executable SCT but an executable eclipse. You have to change that in the scc batch script accordingly.  @jdicks We should mention this in the documentation or let the batch script call the eclipse executable if no SCT executable is present.

stmork, 2016-11-28:
OK, something like
´´´
test -x ./SCT && ./SCT || ./eclipse
´´´
should work.

andreasmuelder, 2016-11-28:
for windows we do not have to change the scc call since we are calling the eclipsec.exe instead of SCT.exe for proper console logging. @tkutz @terfloth How would such a call look like on Macos? Same as Linux?

jdicks, 2016-11-28:
Maybe we should consider this will fail for most custom eclipse based products where the name of the executable might differ.

elgeekomm, 2016-11-28:
Since for some reason I don't have the SCT executable, the following command works for me:

``ECLIPSE_DIR=$(dirname $(readlink -f `which eclipse`))``
``eclipse --launcher.ini ${ECLIPSE_DIR}/headless.ini``

BeckmaR, 2016-11-28:
Hey @elgeekomm, that's a nice idea at least for people who installed eclipse via the system's package manager / added eclipse to their PATH. I assume you do use SCT as a plugin as well instead of a standalone installation? That would be the reason you haven't got a `SCT` binary.

elgeekomm, 2016-11-29:
@BeckmaR correct, we install Eclipse CDT from a shell script, and invoke Eclipse to install all of the plugins we use.

closed 2016-11-29 by andreasmuelder

stmork, 2016-11-29:
I fixed this issue by myself using a short ant script:
```
<project name="SCT generator" default="generate">
	<property file="${user.home}/.sct.properties" />
	<property name="eclipse.home" value="/opt/eclipse/neon-jee"/>

	<target name="generate">
		<exec executable="${eclipse.home}/eclipse" dir=".">
			<arg value="--launcher.ini"/>
			<arg value="${eclipse.home}/headless.ini"/>

			<arg value="-m"/><arg value="model/home.sct"/>
			<arg value="-g"/><arg value="model/home.sgen"/>
			<arg value="-g"/><arg value="model/qt.sgen"/>
		</exec>
	</target>
</project>
```
In `$HOME/.sct.properties` I can define the Eclipse base directory named `eclipse.home` containing the headless builder. Since I know my own infrastructure I assume _eclipse_ as the base runner.

Unfortunately issue #162 is locked to give feedback about the generator itself. It simply runs perfectly well.

andreasmuelder, 2016-12-01:
I just published an intermediate update of the headless generator that fixes this issue.