itemisCREATE/statecharts #1116: Hitting ESC while editing the expression on a transition results in inconsistent behaviour

SimonEit, 2016-11-30:
Reproduction results of the ESC Keypress vary:
- Expression dissappears completely
- Expression text is shown in the model view but not in the properties view
- If you edit an existing expression the ESC keypress functions as an "undo" of your input

andreasmuelder, 2016-12-06:
@SimonEit  I can not reproduce this on Windows. Can you provide detailed steps to reproduce?

SimonEit, 2016-12-06:
@andreasmuelder just did a quick repro at Charalambos Panagiotidis windows machine (with what seemed an older build of the pro version). The behavior was definetly different there: the ESC worked (as expected) as a cancel function and revoked any changes on the transition description. Will try another retest on a current build of the open source version as soon as i find some, who lets me use his windows machine for more than 5 minutes ;-)

SimonEit, 2016-12-06:
@andreasmuelder Okay, i could reproduce the bug on a much smaller scale on windows like this:
- create a **new** transition (seems to only work with new transitions)
- wait for the **automatic popup** of the expression input field
- enter some arbitrary text
- **press ESC** (without "submitting" the text by clicking in the model area)

-> The entered text seems to be submitted (it is shown in the modelling area)
-> A quick look in the properties view shows the text was not submitted (it is nothing there)
-> You can drag around the "ghost text" in the model area
-> If you doubleclick now to edit the text, it disappears (because it was never submitted)

Hope you can reproduce it with these steps

BeckmaR, 2016-12-07:
Thinking about #1117, what exactly do we want to do when the user presses Escape? I would expect it to leave and save my typed text, but that's only me... the german wikipedia says the escape key often leaves and resets text fields, the english wikipedia doesn't say anything like that. Right now I can't find a text field on my computer which respects the Esc key at all.

andreasmuelder, 2016-12-12:
Fixed escape key for new transitions

BeckmaR, 2018-10-17:
This is still an issue (on my Mac).

* When editing a **State**, hitting ESC leaves the state and saves the current contents
* When editing a **transition**, hitting ESC aborts the current edit and deletes it.

There is no key that can always be used to leave the current text field and save the changes. This is quite annoying.