itemisCREATE/statecharts #1117: Result of keypress Enter is not conform with user expectation

SimonEit, 2016-11-30:
Pressing Enter in the model view should result in a submit / safe function of the input.
At the moment it triggers a new line, which is rarely needed and should be remapped to Shift+Enter or Ctrl+Enter

svenjawendler, 2017-05-02:
There is an inconsistency within the definition section and the other textual sections:
Within definition section: RETURN means line break and Strg + RETURN means exit section
Within other textual fields: RETURN means exit and Strg + RETURN means line break

Can this be harmonized?

svenjawendler, 2017-06-26:
This issue was a finding in a Usability Dinner Dortmund/2017.

svenjawendler, 2017-06-26:
This issue is fixed now. Tested with latest build on 26th, June 2017.

closed 2017-06-26 by svenjawendler