itemisCREATE/statecharts #1118: The "New Project"-Wizard should feature a StateChart-Project Folder

SimonEit, 2016-11-30:
At the moment the folder structure of the New Project Wizard suggests to User to chose a Xtend/Java Generator Project, because it is filed under YAKINDU SCT. 

There should be a StateChart-Project (incl. SCTNature). 

It is also difficult to grasp the difference between the selectable Project types. I would recommend to provide a small description of the project in the upper area (under "Select a wizard") of the window. 

[image]

andreasmuelder, 2016-12-06:
@SimonEit The Problem with a generic Statechart Project is that you do not know for which target platform the user wants to generate code for. Depending on this, it should be a JAva or C Project with SCTNature. (Or a simple general project if no code generation is required.) How could we handle this? 3 new project wizards or a wizard with a selection box?

SimonEit, 2016-12-06:
I think the usability issue with this is, that a "newbie" user has a hard time to figure out which project type is suitable for his (beginner) needs. Thus the folder-tree in that wizard should be structured by the goal of use. So i see it more as a descriptive issue. 

Like why are there even Examples/Xtend Examples? Or Plug-In Development? How often does the common user need these? 

Do you see where i am going with this? Otherwise i could provide a quick mockup :)

BeckmaR, 2016-12-07:
I _believe_ the problem here is that this is the standard Eclipse new project wizard, we just hook into it. It would be way more complicated (and for experienced Eclipse users, weird) not to use it.

Stuff like "Examples / Xtend Examples" is there because xtend is one of the installed plugins and it does provide this menu entry. That's not something we could influence.

The only thing we could do is offer a dummy SCT project which in reality is just a "General/Project" as eclipse offers it at the top of the menu with added nature.

It's difficult tgo know how useful this would be, because the regular user should want to develop say Java Code and he will start a Java project and add a statechart later on.

So, all in all, the target audience should be able to understand the way it's done right now.

SimonEit, 2016-12-08:
Just for clarification: the usability finding that triggered this issue came out of the first ux dinner we did in dortmund. Users there were first-time users with little to no eclipse preknowledge with the goal of just modelling and simulating a statechart (no code generation). These users had the problem to finding the "correct" project type.

The question is, if we want to tailor the project wizard to this user group? If not i would agree with @BeckmaR .

terfloth, 2017-01-03:
We also have downloaders and users that are not familiar with Eclipse.

svenjawendler, 2017-06-26:
First time users are confronted with very many menu items / symbols and cannot differ which are important for them and fade out items/symbols that are not needed to work with YAKINDU SCT. For them each item is as important as the other. This 'flood' of items makes it difficult and cumbersome to make progess. But these progresses increase the positive 'user experience', thinking 'YES, I did it. Great tool!' We should remove as much of those stumbling stones as possible, even if it adds little redundancies and somehow provide first time users with blinkers concentrated on YAKINDU SCT. 

Because of this I suggest to add a simple Project beneath YAKINDU SCT: 

[image]

andreasmuelder, 2020-06-10:
We added a new statechart project wizard some time ago

closed 2020-06-10 by andreasmuelder