itemisCREATE/statecharts #1121: NPE when starting simulation

wendehals, 2016-12-05:
Reported against 2.9.0:

When I try to start simulation, an NPE occurs:

!MESSAGE An internal error occurred during: "Launching PoliceCar.sct".
!STACK 0
java.lang.NullPointerException
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer._init_transform_4(DefaultExecutionContextInitializer.java:337)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer._transform(DefaultExecutionContextInitializer.java:323)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer.transform(DefaultExecutionContextInitializer.java:465)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer$7.apply(DefaultExecutionContextInitializer.java:305)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer$7.apply(DefaultExecutionContextInitializer.java:1)
	at org.eclipse.xtext.xbase.lib.IteratorExtensions.forEach(IteratorExtensions.java:363)
	at org.eclipse.xtext.xbase.lib.IterableExtensions.forEach(IterableExtensions.java:333)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer._init_transform_3(DefaultExecutionContextInitializer.java:309)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer._transform(DefaultExecutionContextInitializer.java:285)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer.transform(DefaultExecutionContextInitializer.java:469)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer$1.apply(DefaultExecutionContextInitializer.java:82)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer$1.apply(DefaultExecutionContextInitializer.java:1)
	at org.eclipse.xtext.xbase.lib.IteratorExtensions.forEach(IteratorExtensions.java:363)
	at org.eclipse.xtext.xbase.lib.IterableExtensions.forEach(IterableExtensions.java:333)
	at org.yakindu.sct.simulation.core.sexec.container.DefaultExecutionContextInitializer.initialize(DefaultExecutionContextInitializer.java:86)
	at org.yakindu.sct.simulation.core.sexec.container.AbstractExecutionFlowSimulationEngine.init(AbstractExecutionFlowSimulationEngine.java:118)
	at org.yakindu.sct.simulation.core.debugmodel.SCTDebugTarget.init(SCTDebugTarget.java:65)
	at org.yakindu.sct.simulation.core.launch.AbstractSCTLaunchConfigurationDelegate.launch(AbstractSCTLaunchConfigurationDelegate.java:47)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:885)
	at org.eclipse.debug.internal.core.LaunchConfiguration.launch(LaunchConfiguration.java:739)
	at org.eclipse.debug.internal.ui.DebugUIPlugin.buildAndLaunch(DebugUIPlugin.java:1039)
	at org.eclipse.debug.internal.ui.DebugUIPlugin$8.run(DebugUIPlugin.java:1256)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

BeckmaR, 2016-12-06:
Hey @wendehals, thank you for the report. I allowed myself to format the exception as a code block.

tkutz, 2016-12-06:
Thanks for reporting.
Can you add your model for reproduction?

wendehals, 2016-12-07:
Actually, in a newer installation I can not reproduce the problem. Maybe you can find out by the trace what might have been the problem.

andreasmuelder, 2016-12-08:
Some type could not be inferred, maybe you tried to simulate a faulty model? 
I will close this issue, please feel free to reopen it when the problem is reproducible.

closed 2016-12-08 by andreasmuelder

wendehals, 2016-12-09:
I tried different models, all of them caused the problem. I switched back to 2.8.1 using the same projects and they worked. I guess, there were some problem with the installation.