itemisCREATE/statecharts #1132: Consistent notation(s) for "YAKINDU Statechart Tools"

RainerKlute, 2016-12-14:
Currently there are many different notations for our product and its professional and open source flavours. We should select very few of them and use them consistently throughout the documentation and the software.

Here are some of the notations I saw:
- YAKINDU Statechart Tools
- YAKINDU Statechart Tools Professional Edition
- YAKINDU Statechart Tools Open Edition
- YAKINDU Statechart Tools Pro
- YAKINDU Statechart Tools PRO
- YAKINDU SCT
- SCT

BeckmaR, 2016-12-19:
I'd say we should never use the following:

* YAKINDU Statechart Tools Open Edition
* YAKINDU Statechart Tools Pro
* YAKINDU Statechart Tools PRO
* SCT Pro Edition
* Statechart Tools Pro Edition

RainerKlute, 2016-12-20:
My suggestions:

Full names:
- YAKINDU Statechart Tools
- YAKINDU Statechart Tools Professional Edition

Valid abbreviations, maintaining the 'Y' for brand recognition:
- YAKINDU Statechart Tools Professional
- YSCT
- YSCT PRO

BeckmaR, 2016-12-20:
We have used SCT on so many places and occasions, I believe it would be unwise to drop it now. It's common usage.

RainerKlute, 2016-12-21:
Sometimes you have to slaughter holy cows. And with the release of the professional edition, now is a good point in time to do so and start something new.

andreasmuelder, 2016-12-21:
This will be discussed in the pro repository.

closed 2016-12-21 by andreasmuelder