itemisCREATE/statecharts #1133: Consistent naming of statechart concepts

RainerKlute, 2016-12-14:
Suggestion for a consistent naming of statechart concepts to be use throughout documentation and software:
- a **statechart model** is everything that goes into an _.sct_ file. It consists of:
- the **definition section** and
- the **statechart**, i.e. the graphical part of the statechart model.

terfloth, 2017-01-03:
Ok - we should extend the naming conventions towards the notion of 'state machine'.

Watt is the required action?

andreasmuelder, 2017-01-03:
* a statechart is everything that goes into an .sct file. It consists of:
 * the definition section allows the definition of interfaces/ reactions etc.
 * the graphical canvas contains regions

@RainerKlute Please review the documentation

terfloth, 2017-02-27:
We should also speak of **state machine code** in the context of code generators as well as **state machine execution** in the context of the simulation.

terfloth, 2017-07-20:
@RainerKlute : what is the progress here ?

RainerKlute, 2017-07-20:
Nearing completion, but delayed for more urgent stuff. I don't think it is release relevant.

RainerKlute, 2017-08-05:
Fixed by #1597.

closed 2017-08-31 by andreasmuelder