itemisCREATE/statecharts #1143: Transitions label jumps elsewhere

RainerKlute, 2016-12-21:
Please see the attached model and try to reposition the label of the **Yellow → Red** transition slightly. In most cases the label jumps elsewhere instead of simply remaining where the user dragged it to.

BeckmaR, 2017-01-03:
We'll generally need to redo some parts of diagram drawing, routing, and so on.

Issues related to this:
#29 
#48
#144 
#153 
#360 
#831

RainerKlute, 2017-01-03:
```
oooooo   oooo oooooooooooo  .oooooo..o 
 `888.   .8'  `888'     `8 d8P'    `Y8 
  `888. .8'    888         Y88bo.      
   `888.8'     888oooo8     `"Y8888o.  
    `888'      888    "         `"Y88b 
     888       888       o oo     .d8P 
    o888o     o888ooooood8 8""88888P'  
```

terfloth, 2017-01-03:
This kind of usability issues will be addresses and qualified in the context of the ongoing usability tests. Based on the results further actions will be planned.

svenjawendler, 2017-06-26:
Many participants of the survey and the usability tests so often reported the cumbersome graphical editing. Also today a user reported that he got started very quickly with the tool but stumbled over several issues regarding the graphical editing e.g.:
- he tried to edit a transition but then a popup came up. This happened so often that he got annoyed... the positive aspects of SCT reconciled him especially the prompt feedback on errors and error markes on the failing elements that helped very much.
- during editing YSCT rearanged the elements so often that he spent much time on layouting.

I presented a tutorial to a new user. He also complained about the layouting... 

Is it possible to enhance this behavior?

RainerKlute, 2017-06-26:
This is also a long-standing complaint of mine, so I whole-heartedly agree!

tkutz, 2017-08-21:
[edge label behavor.mov.zip](https://github.com/Yakindu/statecharts/files/1238253/edge.label.behavor.mov.zip)

commit 4270cd97 (2018-02-07): Merge pull request #1892 from Yakindu/issue-1143-transition-jump

fixes #1143 the transition label jumping mentioned in
commit 93aa123b (2018-02-07): fixes #1143 the transition label jumping mentioned in

reused sirius label routing (well done @ obeo)
closed 2018-02-07 by RainerKlute