itemisCREATE/statecharts #1144: #define to save space

robfox1976, 2016-12-23:
I'm working on my first project using Yakindu. Although I've found it very useful, I've come to a point where I've run out of room on my micro!

One thing that I would like to do, is rather than use "var readonly" I'd ideally like to use a #define.

I've had a look around but I couldn't see anything.

I did try changing all my "var readonly" variables to "const", but one issue I had with this is that they were all under "internal". When I changed all to "const" I got an error "expected a declaration", as I basically had 

`/*! Type definition of the data structure for the TestMInternal interface scope. */
typedef struct
{

} TestMInternal;`

BeckmaR, 2017-01-03:
In the open source edition of YAKINDU SCT, it is not possible to use a `#define`. Did you already try the Inlining-Features of the Code generator, compiler optimizations and switching to the default build instead of debug?

robfox1976, 2017-01-04:
I have not tried the inlining features yet. I have set the optimisation settings to high for size and this has helped considerably.

Are there any plans to introduce something like #define?

BeckmaR, 2017-02-17:
You can only use defines from header files in the Professional Edition using our Deep C Integration. It allows you to use all types, variables and defines from your header files.

The bug you mentioned in your issue description concerning the empty structs has been fixed recently with PR #1196, and we plan to release a new version including this fix on next friday.

closed 2018-01-11 by andreasmuelder