itemisCREATE/statecharts #1146: Job: "Writing new resource descriptions chunk i of n" blocks Eclipse

lostwald, 2017-01-02:
Hi,

After installation of the the Yakindu SCT Pro Versionw ith c integration several users have the problem, that a job is running (for several days). The Job Description is "Writing new resource descriptions chunk i of n (see screenshot). The problem disappear after deinstalling Yakindu SCT.

lostwald, 2017-01-02:
Additional information,

when this job is started it is not possible to access types defined in C Domain source code.

The job finally quits whith an out of memory exception after consuming more than 16GB heap space.

tkutz, 2017-01-02:
Did you only uninstall the Pro edition? So is it a problem in the Pro Edition only, or also in the Open Edition of SCT?

lostwald, 2017-01-02:
The problem is in the Pro edition only. I can reproduce the problem by opening a project in the 
Project Explorer with some "bigger" amount of source code. It does not occur when only some trivial c-files with types are available.

paphko, 2017-01-02:
I found the culprit: https://github.com/eclipse/xtext-eclipse/blob/master/org.eclipse.xtext.builder/src/org/eclipse/xtext/builder/clustering/ClusteringBuilderState.java#L216

lostwald, 2017-01-03:
Hi,
but the message is "Writing new resource descriptions chunk", instead of "Updating resource descriptions chunk " like in line https://github.com/eclipse/xtext-eclipse/blob/master/org.eclipse.xtext.builder/src/org/eclipse/xtext/builder/clustering/ClusteringBuilderState.java#L386

andreasmuelder, 2017-01-04:
Currently, all header files from the whole workspace are indexed during a workspace build. We will change this implementation so that only header files that are imported into a statechart will be indexed. 
There is also a performance issue when using a lot of defines, maybe this is the case in your example. 
This issue will be closed and reopened in the internal professional edition repository.

closed 2017-01-04 by andreasmuelder