itemisCREATE/statecharts #1151: Move statechart specification to collapsable editor area just like the palette

paphko, 2017-01-03:
The specification of a statechart may contain a lot of text which is not really a graphical part of the statechart. It looks especially awkward when it interfers with the graphical part as shown here: 

[image]

I would suggest to add a collapsable editor which is by default located at the left-hand side of the graphical editor. Then the specification can be shown and also resized as needed, it does not clutter the graphical representation of the statechart, and syntax highlighting could be enabled permanently.

BeckmaR, 2017-01-03:
I like this idea. I don't think the specification needs to be always visible. This would also enable us to show the specification in sub-statecharts.

But one tip for now: You can move and resize regions and the specification section. It does not need to overlap with your region.

andreasmuelder, 2017-01-23:
We should plan this for one of the next milestones then

terfloth, 2017-02-27:
That seems to make sense > discuss with usability engineer

svenjawendler, 2017-05-02:
To adapt the UI to the personal needs is important to enhance usability. To fade out features that are currently not important enables the user to focus. @SimonEit Do you agree?

svenjawendler, 2017-05-08:
The survey participants also mentioned this point multiple times. Here are some cites: separation of textual FSM content definition from graphical view into own file; Why is the statechart description language written in a field in the .sct model and not simply in a distinct file (e.g. model.sdl)?; The input for the events, variables and so on should not be in the drawing area.; Provide a high-level textual script language to edit diagrams; Text only representation of state machines. Graphical representation is automatically generated from the DSL representation of the state machine.

andreasmuelder, 2017-05-09:
https://github.com/Yakindu/statecharts/issues/849 is referenced

terfloth, 2017-07-10:
Sketch proposal of solution.

robert-rudi, 2017-11-29:
As implemented in #1740 there is a new feature, where the inlined definition section can be pinned to the left side of the statechart diagram editor. With the options to collapse and expand the pinned definition section, the behaviour is as the one of the palette on the right side. 

As a result of a short usability discussion it should be evaluated, if the option to re-inline the definition section is needed at all. The only argument we have found to switch to the old definition section appearance, is the printing-feature for the statechart diagram. In case the user wants to print the diagram, he wouldn't get the pinned definition section. 

As an internal feature request there might be the possibility to print the pinned definition section besides the statechart diagram when exporting the diagram to an image with the new ImageRenderer (in a diagram: Right Click -> File -> Save As Image File ... ).
The complexity for this request must be evaluated.

PR #1740 (2017-12-11): Issue statecharteditor prototype
Fix #1347 
Fix #1151
closed 2017-12-11 by andreasmuelder