itemisCREATE/statecharts #1167: scc does not work with Eclipse neon.2

stmork, 2017-01-18:
When starting scc using Linux I get the following dialog:
[image]
After clicking the button the following message appears in the command shell:
```
Launching /tmp/eclipse-jee/eclipse...
plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.1.400.v20160518-1444: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden
```
With Eclipse neon.1a everything works fine.

stmork, 2017-01-18:
I recognized that removing the following lines from the _headless.ini_ helps:
```
-startup
plugins/org.eclipse.equinox.launcher_1.3.200.v20160318-1642.jar
--launcher.library
plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.1.400.v20160518-1444
--launcher.XXMaxPermSize
512M
```

jdicks, 2017-01-18:
Hi @stmork, 
thanks for reporting - we observed similar behaviour and already fixed it using the same solution you mentioned for win and linux. ;) 
For Mac it seems to be not that easy - might be caused by the internal .app structure and we have to investigate further.

We might keep the XXMaxPermSize property as reasonable compromise a bit longer because we still support java 7 environments where sometimes such small things make everything a bit more easy for users don't want to get into eclipse that much. We hope you can live with the warning on startup of a java 8 vm a bit longer or just remove it from your local installation.

You can expect the launcher issue to be fixed by the upcomming version of the pro edition (pro updatesite) for win and linux - hopefully for mac also. 

I will close this issue - feel free to reopen if there are any questions left.

Regards!

closed 2017-01-18 by jdicks