itemisCREATE/statecharts #1168: Bug in Doku ( ??? instead of an arrow)

DominikTesch, 2017-01-19:
[image]

RainerKlute, 2017-01-23:
I checked the generated HTML code for both Eclipse help and website, and it is perfectly okay.

RainerKlute, 2017-01-23:
Ah, okay! While as [already stated](https://github.com/Yakindu/statecharts/issues/1168#issuecomment-274458328) the generated HTML code is fine, this is only true in my development environment, but **not** in the built product. Somewhere in the toolchain all special characters got replaced by "???".

RainerKlute, 2017-01-23:
In older versions this worked fine, but has been broken somewhere between 2.6.0 and 2.9.1.

andreasmuelder, 2017-01-23:
Reason is a wrong encoding on the new build slaves, it should be UTF-8. fixed on build server

closed 2017-01-23 by andreasmuelder