itemisCREATE/statecharts #1170: Exporting a statechart as an image file via Generator Model

lostwald, 2017-01-23:
Hi,

Yakindu SCT has a nice feature to export a statechart to a image file. We link to the exports from our software architecture and this includes the description of the statechart very well.

However doing this export manually often leads to inconsistency: the model gets updated but it is forgotten to update the "export to image file" as well. Is it possible to include the export as a generation step each time the gode generation is executed (e.g. via the generator model)?

Kind regards,
Lars

andreasmuelder, 2017-01-23:
This is somehow related to https://github.com/Yakindu/statecharts/issues/1135

andreasmuelder, 2018-01-02:
This feature is available in the upcoming SCT release, a SCT image generator can be configured for automatically export images.

closed 2018-01-02 by andreasmuelder