itemisCREATE/statecharts #1177: Canvas should have more margin

BeckmaR, 2017-01-24:
When working with multiple regions, the drawing canvas should expand more to give some slack to the drawing area. Consider the following screenshot:

[image]

it is not possible to scroll down any more and working like this feels cramped. At least half the height of that region should be added in the bottom and the right of the canvas, maybe more.

BeckmaR, 2017-01-24:
Assigned @SimonEit for a statement because it is a usability issue.

SimonEit, 2017-09-08:
Agree with @BeckmaR that the margin around regions should be a bit bigger. Would rate this issue as very minor convenience feature though.