itemisCREATE/statecharts #1188: Empty struct generated from interface description

lostwald, 2017-02-06:
Hi,

the following interface description generates a empty struct:

[image]

[image]

terfloth, 2017-02-06:
Seems to be identical to of #596. This has to be qualified.

BeckmaR, 2017-02-08:
Hey @lostwald, this should have been fixed in release 2.7.1, according to the commit history (#869 merged on August 3rd 2016). Could you please verify that the version used here was not older than that?

BeckmaR, 2017-02-08:
I can verify this issue still exists.

This statechart:
[image]
produces this empty interface struct in `OSM.h`:
```
/*! Type definition of the data structure for the OSMInternal interface scope. */
typedef struct
{
} OSMInternal;
```
Instead, the following line is added to `OSM.c`:
```
const sc_integer OSM_OSMINTERNAL_OSM_STATE_INIT = 0;
```
The code seems to be referencing this constant correctly. A getter-function is not generated.

If the `const` is moved to an interface, the same happens - an empty struct, a const in `OSM.c`, and a getter function is generated. Trying to access the consts directly obviously does not work, because the symbol is not known.

Basically, it could be worse, everything works, only the code style is bad.

PR #1196 (2017-02-15): Remove void and const from typeRelevantDeclarations
Just removes the types that are const or void from typeRelevantDeclarations, so that .empty actually returns that it will generate nothing if it won't.

Fixes #1188
closed 2017-02-15 by andreasmuelder