itemisCREATE/statecharts #1194: Improve cancelation of SCTValidationJob

jdicks, 2017-02-09:
# Description:
In SCT the editor use a the SCTValidationJob to validate the resource content.
The job is scheduled by the ResourceSetValidationListener which is a ResourceListener which is notified after each execution of a transaction (see EMF transaction).

Before new scheduling of the job, a potential already running job is cancelled. This should ensure the liveness and if the user do fast interactions with the editor.

This implementation seems to work fine as long as you work with small models. But it actually does not work.
This gets obvious if you work with large models, where the validation job runs longer than some milliseconds.
Than the validation job actually blocks the user interaction.

# Root cause:
The location to cancel a running job is wrong.
The job should be cancelled after a new notification indicates a new subsequent change of the model. But because of the locking of the editing domain during the validation it is not possible to create a new change which change the model and leads to such a new notification.

# Possible solution:
It is necessary to ensure, that the running job is actually cancelled before a user interaction leads to asking for a new lock on the editing domain.
For this purpose the only location which come into my mind is the TransactionalCommandStack. This is a central location where each user interaction is going through. So in the methods execute, undo and redo, a cancel and reschedule of the job will solve the problem.

# Additional Info: 
I found out, that in GMF use case the canceling of the validation job on executing a command in the command stack like described in this bug is not sufficient.
GMF very very often calls org.eclipse.emf.transaction.TransactionalEditingDomain.runExclusive(Runnable) which also blocks the ui thread until read only access to the model could be granted.
So also on this case the validation job must be canceled. Anyway it is much more tricky to get a good UX here, because this canceling happens really very often in GMF. So it could be easily the case, that a validation which really takes long may never ends. Also validation may only be rescheduled if really was canceled beforhand...

jdicks, 2017-02-09:
attached a prototypical patch the StatechartDiagramEditor:
[StateMachineEditor.txt](https://github.com/Yakindu/statecharts/files/764118/StateMachineEditor.txt)

andreasmuelder, 2017-03-03:
I propose to not execute the Validation Job within a runexclusive on the TransactionalEditingDomain, see #1249 The validation job does not block UI interaction and allows to change the semantic model during validation. If this happens, the Job gets canceled and restarted.

andreasmuelder, 2017-03-21:
Second approach committed to branch is a ShadowModelValidation Job that clones the whole resource before validation und validates on the shadow model without a read transaction. More time and memory consuming but way more responsive  UI.

andreasmuelder, 2017-03-21:
With PR #1201 the new default validation job uses a shadow model for vaidation.

closed 2017-03-21 by andreasmuelder