itemisCREATE/statecharts #1197: Provide hint for unsupported parameters in SCC

jdicks, 2017-02-15:
For 2.9.2 we removed the sgen parameter from headless code generator infrastructure, because the -m parameter was sufficient to load all kind of model files needed to generate code out of sct files.

This change broke the build system of some users and our CLI does not give a proper hint about this change.

We should add a more strict parameter handling and give a proper msg if violations occur.
Further we should handle the case where files can't be found more explicit:
* avoid stacktrace
* maybe print an absolute path

See discussion in https://groups.google.com/forum/?fromgroups=#!topic/yakindu-user/HH7yK3swh2A

tkutz, 2017-02-15:
We should also consider making parameters required if necessary. The problem here is, that the parser should not check requiredness if the help argument is used (-h). For this we would need to mutually exclusive option groups as described in this post:

http://stackoverflow.com/questions/31386110/conditional-command-line-option-with-apache-commons-cli

terfloth, 2017-02-27:
Command line options are API so we should handle changes carefully. I see the following phases:

1. normal
1. deprecated
1. unsupported

A deprecated option can be used but should result in a warning.
An unsupported option should result in an error stating 'option xyz not supported' 

Add test cases for unsupported options to the test plan.

andreasmuelder, 2017-04-06:
@jdicks @terfloth @tkutz Since the parameter will stay the same in the future I will close this as invalid.

closed 2017-04-06 by andreasmuelder