itemisCREATE/statecharts #1199: StackOverflowError while editing a State Behaviour

SimonEit, 2017-02-16:
I found a reproducable Bug while editing state behaviours:

Operating System: **Mac OS 10.12.3 Sierra**
SCT Version: **YAKINDU Statechart Tools PRO - Product 1.0.0.201612161424**

**Error Message:**
[image]
[image]

**Steps to reproduce:**
Edit a State Behaviour either in the graphical editor or via properties view. After a couple characters of input the error message pops up.

**-> It is possible to continue working after dismissing the error message.**

BeckmaR, 2017-02-16:
Can't reproduce right now on Ubuntu 16.04 64 bit. Can you upload the project please?

SimonEit, 2017-02-16:
[test_2.zip](https://github.com/Yakindu/statecharts/files/780091/test_2.zip)

This zip file cointains the project, which causes the error. 

The ExceptionstackTrace is pretty long, hence the .log and .logbak files got quite large :)

SimonEit, 2017-02-16:
After further investigation it seems to be limited on this workspace / project only. It is not reproducable outside of this scope. So our current guess is, that it is a weird / rare combination of circumstances that cause the problem. 

If i manage to reproduce the error in another scope, i will report back.

andreasmuelder, 2017-02-24:
The same error was reported via OTRS. We should investigate on this.

BeckmaR, 2017-02-27:
[The logfile from @SimonEit ](https://github.com/Yakindu/statecharts/files/803877/log.txt)
The start of the recursion:
```
java.lang.StackOverflowError
	at org.eclipse.emf.common.util.SegmentSequence.matches(SegmentSequence.java:524)
	at org.eclipse.emf.common.util.SegmentSequence$SegmentSequencePool$StringAccessUnit.matches(SegmentSequence.java:2160)
	at org.eclipse.emf.common.util.SegmentSequence$SegmentSequencePool$StringAccessUnit.matches(SegmentSequence.java:1)
	at org.eclipse.emf.common.util.Pool$AccessUnit.match(Pool.java:255)
	at org.eclipse.emf.common.util.Pool.doIntern(Pool.java:924)
	at org.eclipse.emf.common.util.SegmentSequence$SegmentSequencePool.intern(SegmentSequence.java:2664)
	at org.eclipse.emf.common.util.SegmentSequence.create(SegmentSequence.java:2715)
	at org.eclipse.emf.ecore.resource.impl.ResourceImpl.getEObject(ResourceImpl.java:742)
	at org.eclipse.xtext.resource.XtextResource.access$101(XtextResource.java:63)
	at org.eclipse.xtext.resource.XtextResource$1.getEObject(XtextResource.java:119)
	at org.eclipse.xtext.resource.DefaultFragmentProvider.getEObject(DefaultFragmentProvider.java:28)
	at org.eclipse.xtext.resource.XtextResource.basicGetEObject(XtextResource.java:363)
	at org.eclipse.xtext.resource.XtextResource.getEObject(XtextResource.java:349)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.getEObject(LazyLinkingResource.java:232)
	at org.eclipse.emf.ecore.resource.impl.ResourceSetImpl.getEObject(ResourceSetImpl.java:223)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:199)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:259)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eResolveProxy(BasicEObjectImpl.java:1477)
	at org.eclipse.emf.ecore.util.EcoreEList.resolveProxy(EcoreEList.java:206)
	at org.eclipse.emf.ecore.util.EcoreEList.resolve(EcoreEList.java:161)
	at org.eclipse.emf.ecore.util.EObjectContainmentEList$Resolving.resolve(EObjectContainmentEList.java:105)
	at org.eclipse.emf.common.util.BasicEList.get(BasicEList.java:348)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eObjectForURIFragmentSegment(BasicEObjectImpl.java:539)
	at org.eclipse.emf.ecore.resource.impl.ResourceImpl.getEObject(ResourceImpl.java:766)
```
After that, the following part repeats:
```
at org.eclipse.emf.ecore.resource.impl.ResourceImpl.getEObject(ResourceImpl.java:742)
	at org.eclipse.xtext.resource.XtextResource.access$101(XtextResource.java:63)
	at org.eclipse.xtext.resource.XtextResource$1.getEObject(XtextResource.java:119)
	at org.eclipse.xtext.resource.DefaultFragmentProvider.getEObject(DefaultFragmentProvider.java:28)
	at org.eclipse.xtext.resource.XtextResource.basicGetEObject(XtextResource.java:363)
	at org.eclipse.xtext.resource.XtextResource.getEObject(XtextResource.java:349)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.getEObject(LazyLinkingResource.java:232)
	at org.eclipse.emf.ecore.resource.impl.ResourceSetImpl.getEObject(ResourceSetImpl.java:223)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:199)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:259)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eResolveProxy(BasicEObjectImpl.java:1477)
	at org.eclipse.emf.ecore.util.EcoreEList.resolveProxy(EcoreEList.java:206)
	at org.eclipse.emf.ecore.util.EcoreEList.resolve(EcoreEList.java:161)
	at org.eclipse.emf.ecore.util.EObjectContainmentEList$Resolving.resolve(EObjectContainmentEList.java:105)
	at org.eclipse.emf.common.util.BasicEList.get(BasicEList.java:348)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eObjectForURIFragmentSegment(BasicEObjectImpl.java:539)
	at org.eclipse.emf.ecore.resource.impl.ResourceImpl.getEObject(ResourceImpl.java:766)
```

andreasmuelder, 2017-02-27:
This only happens when using Xtext 2.11

andreasmuelder, 2017-02-28:
The newly introduced default binding to org.eclipse.xtext.parser.antlr.IReferableElementsUnloader 
in Xtext 2.11 causes this Stackoverflow error, see https://github.com/eclipse/xtext-core/issues/122
and the resulting PR https://github.com/eclipse/xtext-core/pull/142

cdietrich, 2017-02-28:
@andreasmuelder can you share some more hints. when and why is unload called in a way that this leads to follow up problems?

andreasmuelder, 2017-02-28:
@cdietrich the interesting part is PartialParsingHelper line 156 - 166:
if (feature.isMany()) {
	List featureValueList = (List) oldSemanticParentElement.eGet(feature);
	int index = featureValueList.indexOf(oldSemanticElement);
	unloadSemanticObject(oldSemanticElement);
	EObject newSemanticObject = newParseResult.getRootASTElement();
	if (newSemanticObject != null) {
		featureValueList.set(index, newParseResult.getRootASTElement());
	} else {
		featureValueList.remove(index);
	}
}
What happens when a unloader is bound is that the featureValueList as part of the oldSemanticElement becomes a proxy.  FeatureValueList.set is called on a proxy then what leads somehow to the stack trace above.

andreasmuelder, 2017-02-28:
I am closing this here since we have merged a Bugfix for SCT. We can remove the binding when the root cause is fixed in the upcoming xtext release.

closed 2017-02-28 by andreasmuelder