itemisCREATE/statecharts #1206: Path to Headers relative

DefCon-CC, 2017-02-21:
I observed that in Yakindu SCT 2.9.2. (and prior) the Path to Header-Files will be written relative into the Yakindu SCT generated files. This causes problems with the Software Unit-Test Concept of our Company, where all Header-Files are expected to be in one Folder. A option to switch off relative paths would be nice!

[image]

tkutz, 2017-02-21:
So do I understand it correctly, specifying the same folder for sc_types.h in the generator model is not an option because you share the file among multiple projects?

andreasmuelder, 2017-02-21:
This change was introduced with https://github.com/Yakindu/statecharts/issues/627

DefCon-CC, 2017-02-21:
No, the problem ist that a script in our Componay fetches all Header-Files in our Project and copys it to one Folder. This folder will be provided to the Software Unit-Test Tool. The Unit-Test Tool expects all Header-Files in this folder; we have no subfolders. When there are relative paths to Header-Files in Code the Unit-Test Toll will not find it.

tkutz, 2017-02-21:
So why don't you specify the same folder for sc_types.h and the statemachine headers in the generator model?

DefCon-CC, 2017-02-21:
Because we have drawn several state machines in different folders in our Project, but they shall share all the same sc_types.h

tkutz, 2017-02-22:
Okay, we will probably provide a new generator option to switch from relative paths to simple includes. So you would have the following include:

````
#include "sc_types.h"
````
Since we are already in our QA phase, we won't add that feature for this week's release but for the upcoming one.

DefCon-CC, 2017-02-22:
Sounds good :-)

terfloth, 2017-02-23:
@DefCon-CC -  If i understand you right, then you add the directory of sc_type.h to the include path when invoking the compiler?

DefCon-CC, 2017-02-24:
@terfloth For compiling we have a script which fetches the path to every source- and header-File, these paths are written into a file which will be invoked by the compiler

DefCon-CC, 2017-05-16:
Is this feature part of Yakindu SCT 2.9.3?

terfloth, 2017-05-17:
The feature is merged is available on the master. It will be part of release 2.9.4 or 2.10.0 or 3.0.0 depending on how we name the next release ;-). 

it is also included in our latest build on the milestones update site (as 2.9.3)

DefCon-CC, 2017-05-18:
I've tested it and it seems to work. Thx!