itemisCREATE/statecharts #1240: Null Pointer exception when generating Statechart

DefCon-CC, 2017-03-01:
I'm expirencing a Java Null pointer Exception in Yakindu SCT 29.2 when generating code from the Statechart. The Null Pointer is vanished after restart of Eclipse!

Generating 'RailAmtSel' to target project 'Appl' ...
java.lang.NullPointerException
	at org.yakindu.sct.generator.c.ExpressionCode._code(ExpressionCode.java:396)
	at org.yakindu.sct.generator.c.ExpressionCode.code(ExpressionCode.java:534)
	at org.yakindu.sct.generator.c.ExpressionCode._code(ExpressionCode.java:465)
	at org.yakindu.sct.generator.c.ExpressionCode.code(ExpressionCode.java:552)
	at org.yakindu.sct.generator.core.templates.Expressions._code(Expressions.java:37)
	at org.yakindu.sct.generator.c.ExpressionCode.code(ExpressionCode.java:538)
	at org.yakindu.sct.generator.c.ExpressionCode._sc_boolean_code(ExpressionCode.java:504)
	at org.yakindu.sct.generator.c.ExpressionCode.sc_boolean_code(ExpressionCode.java:619)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:409)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:540)
	at org.yakindu.sct.generator.c.StatemachineSource._functionImplementation(StatemachineSource.java:1259)
	at org.yakindu.sct.generator.c.StatemachineSource.functionImplementation(StatemachineSource.java:1307)
	at org.yakindu.sct.generator.c.StatemachineSource.toImplementation(StatemachineSource.java:1234)
	at org.yakindu.sct.generator.c.StatemachineSource.functionImplementations(StatemachineSource.java:1199)
	at org.yakindu.sct.generator.c.StatemachineSource.content(StatemachineSource.java:207)
	at org.yakindu.sct.generator.c.CGenerator.generateArtifacts(CGenerator.java:89)
	at org.yakindu.sct.generator.c.CGenerator.generate(CGenerator.java:80)
	at org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor.execute(SExecGeneratorEntryExecutor.java:54)
	at org.yakindu.sct.generator.core.execution.AbstractGeneratorEntryExecutor.execute(AbstractGeneratorEntryExecutor.java:58)
	at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.execute(GeneratorExecutorLookup.java:51)
	at org.yakindu.sct.generator.builder.EclipseContextGeneratorExecutorLookup$1.run(EclipseContextGeneratorExecutorLookup.java:43)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
RailAmtSel done.

BeckmaR, 2017-03-01:
Hey @DefCon-CC,
thank you for the report. So basically when you restart Eclipse / SCT the bug does not appear again when you try to generate once more?

DefCon-CC, 2017-03-01:
@BeckmaR Yes correct, the bug disappears after restart of Eclipse / SCT

BeckmaR, 2017-03-01:
Are you able to provide an example project in which the error occurs?

DefCon-CC, 2017-03-01:
Unfortunately not. I'm not able to reproduce the error anymore.

andreasmuelder, 2017-05-08:
Closing this as not reproduceable.

andreasmuelder, 2017-05-08:
Please reopen if this error occures again.

closed 2017-05-08 by andreasmuelder