itemisCREATE/statecharts #1241: Size of Choice Symbol not changeable

DefCon-CC, 2017-03-01:
It would be great if the size of a "Choice" Symbol would changeable, because it is hard to click on the "Choice" Symbol when many lines and arrows are connected to it.

[image]

andreasmuelder, 2017-03-01:
This is related to https://github.com/Yakindu/statecharts/issues/432

BeckmaR, 2017-03-01:
I have got one tip for you: go through the incoming transitions, check out the properties tab in the bottom and change the arrow routing from "oblique" to "rectilinear". Then you will have it easier to align the transitions on top of each other and clean up the statechart a little. This makes handling the routing of transitions a bit easier in general. We changed the default setting in the last release, but it only applies to new transitions.

PR #1351 (2017-05-05): The choice can now be resized with fix aspect ratio
This fixes #1241 and #432
commit 8a9ccb5c (2017-05-05): The choice can now be resized with fix aspect ratio (#1351)

This fixes #1241 and #432
closed 2017-05-05 by andreasmuelder