itemisCREATE/statecharts #1250: FMI / FMU code generation

lionelbelmon, 2017-03-06:
It would be very useful to directly package the state chart to a FMU.
see https://fmi-standard.org/.
https://fmi-standard.org/tools

The standard is supported by >100 simulation tools. This would allow to directly load the state chart in the simulation tool and couple it other simulation models.

terfloth, 2017-03-09:
Yes - we are aware of FMI and it would fit well to SCT. Due to lots of current development topics we won't be able to schedule this feature request within the next months. So i move it to the backlog until it becomes more relevant or some interested party is willing to sponsor the feature development.

lionelbelmon, 2017-03-13:
well... having a generation of FMUs from the STC would change a lot the use-case and open it to a new community of users, which are up to now unaware of the tool I believe.  These people are usually not "coders" or "software engineers" but are "tool users". Generating a FMU and the dll would hide all/most code complexity and tricks needed to integrate the generated code inside a user-software.

tbeu, 2018-08-24:
Is this related to https://twitter.com/yakindu/status/1029694071311425536?

terfloth, 2018-08-28:
Yes - it is. 

We re currently prototyping the FMU generation from statecharts. We expect to provide a first version of this feature during the last quarter of this year.

andreasmuelder, 2018-08-28:
I did some clean up only 72 branches left :)

terfloth, 2020-11-26:
A FMU generator is available on request. It will soon be provided on the public YAKINDU-Labs update site.

terfloth, 2020-12-30:
@lionelbelmon @tbeu We finally support FMU export for FMI Co-Simulation 2.0. The required code generator is provided for the current 4.0.4 release of YAKINDU SCT and available on our labs update site. Please check our  [user gude](https://www.itemis.com/en/yakindu/state-machine/documentation/user-guide/lab_projects_fmu_codegen#lab_projects_fmu_codegen) for details. YAKINDU is also already listed on the [FMI standards tool page](https://fmi-standard.org/tools/) 

Currently we are working on completing the FMU generator and appreciate any feedback. So if you want to take a look at the FMU export don't hesitate to contact use here or via email to statecharts@itemis.com .

closed 2021-01-04 by terfloth