itemisCREATE/statecharts #1256: Better Documentation index

andreasmuelder, 2017-03-07:
Use a more fine grained index for the documentation on our website.

terfloth, 2017-05-04:
First evaluate options.

RainerKlute, 2017-05-05:
The most important information is that Textile by itself does neither support an index nor a table of contents (TOC). Our options:
- Investigate what Mylyn Wikitext offers. If needed, we should also investigate developer versions.
- Extend Mylyn by a so-called "language customization".
- Look at solutions developed by others – probably not very promising.
- Develop our own solution that parses a couple of Textile files and emits index and/or TOC.
- Replace Textile by something better, e.g. DocBook or AsciiDoc.

[image]

RainerKlute, 2017-05-10:
Using the Mylyn Wikitext API looks very promising. Writing a custom document builder producing a table of contents artifact is definitely possible.

RainerKlute, 2017-08-03:
Thomas Laß provided a layout draft.

RainerKlute, 2017-09-22:
Moving the index to the left-hand side requires non-trivial changes to the new Textile-to-Hubspot generator.

RainerKlute, 2017-10-23:
Fixed with pull request #1688.

closed 2017-10-23 by RainerKlute