itemisCREATE/statecharts #1264: Useless exceptions thrown by scc, maybe related to GUI classes being loaded

BeckmaR, 2017-03-15:
Liudmila Dobriakova just posted [this](https://groups.google.com/forum/#!topic/yakindu-user/3ctlLV0vaQY) into our user group:

```
Launching /home/yakindu-sctpro/SCT...
OpenJDK 64-Bit Server VM warning: ignoring option PermSize=256m; support was rem                                                                                 oved in 8.0
--------------------------------------------------------
YAKINDU Statechart Tools Headless Generator ((c) by itemis AG)

          Visit http://www.statecharts.org
--------------------------------------------------------
Using base dir: /home/Dem1/eu.hyvar.metadata.deltaecore.example
Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Unable to init server: Could not connect: Connection refused
Error loading resource '/home/Dem1/eu.hyvar.metadata.deltaecore.example/src/sc_types.h': com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: com.google.inject.internal.util.$ComputationException: java.lang.NoClassDefFoundError: org/eclipse/ui/IFileEditorInput
Error loading resource ....
```
```
40 model(s) loaded.
Generating to : /home/Dem1
Generating 'eCall' to target project 'eu.hyvar.metadata.deltaecore.example' ...
eCall done.
Generated (1 of 1) gen model(s).
```
Mir is the new display server on Ubuntu which is not yet ready for production. It looks like the headless code generator would still load some GUI classes (`org/eclipse/ui/IFileEditorInput`) and Mir seems to have problems still when gui stuff is called from the terminal for whatever reason.
Code generation seems to have worked successfully, but the amount of error messages that gets thrown sometimes is of course an annoyance.

BeckmaR, 2017-03-22:
We got a full stacktrace:
[text.txt](https://github.com/Yakindu/statecharts/files/861870/text.txt)

BeckmaR, 2017-03-30:
So the full list of classes and interfaces that get thrown through the Injector and finally lead to the NoCLassDefFoundError: IFileEditorInput:

* `DefaultRefactoringDocumentProvider`
* `IRefactoringDocument`
* `RefactoringUpdateAcceptor`
* `IRefactoringUpdateAcceptor`
* `RenameEllementProcessor` (`updateAcceptorProvider`)
* `AbstractRenameProcessor` 
* `DefaultRenameRefactoringProvider`
* `IRenameRefactoringProvider`

all this is loaded because `org.yakindu.sct.resource` is loaded via injection from `CHeaderResourceFactory`

BeckmaR, 2017-03-30:
The error is related to the fact that a `sc_types.h` exists that was generated before. It is loaded as a model by the headless generator, and then the C Domain is brought up to handle this file (used here is a pro edition of SCT, but the statechart does not use pro features) and everything goes wrong.

BeckmaR, 2017-03-30:
The issue here is that a `sc_types` exists, but because the statechart is of the default domain, the Guice bindings are not modified to use `HeadlessCHeaderResourceFactory`for c files.

BeckmaR, 2017-03-31:
The original poster from the user group reported that it helped her to provide her docker image in which scc runs with another environment variable:
```
docker run...  -e DISPLAY=unix$DISPLAY ...
```
This works around the problem that we inject a class that has UI dependencies down the road which fail because the docker image doesn't seem to define the $DISPLAY. This fix did **not** work for me on the provided docker image.

andreasmuelder, 2017-04-04:
It seems that the XTextBuilder is running, we should disable the auto build during headless execution.
...
final IWorkspaceDescription desc = workspace.getDescription();
desc.setAutoBuilding(false);

Maybe we should restore the original value after headless execution.

jdicks, 2017-04-13:
@BeckmaR & @Miliks thanks for analysis and https://github.com/Miliks/HeadlessStatechart. we improved the following: 
* disable auto-build during headless execution explicitly
* load domain specific models (e.g. *.h) just for domains which are used in the given set of SCT models 
  * e.g. *.h will be loaded if c-domain is used

will be part of the next release.

closed 2017-04-13 by jdicks