itemisCREATE/statecharts #1267: Create a tutorial series to cater to first time users

SimonEit, 2017-03-21:
The concept for this is to built a very simple tutorial project series that contains tasks, examples and sample solutions for the target audience of 'first time users' who have a cold start with the tool.

The goal is not to educate the users about the basic principle of statemachines though, but is focused on the functionality of the tool.

terfloth, 2017-03-27:
@SimonEit : what is the current status?

svenjawendler, 2017-03-27:
Maybe we could create cheatsheets on this?

svenjawendler, 2017-06-12:
Result 1: An example project with some very basic lessons. It needs a review: https://drive.google.com/drive/folders/0BxqrVB1ADkfARG16R25uckc2U00 
Result 2: To make cheatsheets work successfully with YAKINDU Statechart Tools we need more COMMANDS. Currently there is only the COMMAND: Create Subdiagramm. We need Commands for create state, create transition.

andreasmuelder, 2017-06-27:
moving to backlog

andreasmuelder, 2018-01-29:
closing as orphaned

closed 2018-01-29 by andreasmuelder