itemisCREATE/statecharts #1275: Add a checkbox "save automatically" to the save dialog when running a simulation

SimonEit, 2017-03-22:
At the moment the dialog pops up every time you run a simulation on an unsaved statechart. Far more often the user wants to save the model at this point rather than simulating the model in a dirty state. Therefore the dialog should contain a checkbox, which provides an auto-save functionality.

svenjawendler, 2017-05-02:
[image]

svenjawendler, 2017-05-02:
SOLUTION

It can be done by setting the preference:
[image]

Maybe we could ship this with the YSCT preferences?
/instance/org.eclipse.debug.ui/org.eclipse.debug.ui.save_dirty_editors_before_launch=always

andreasmuelder, 2017-05-05:
I think the problem here is that a different dialog shows up when a sct editor is in dirty state

![unbenannt](https://cloud.githubusercontent.com/assets/6338708/25741527/79b065d4-318b-
11e7-8bb1-0a4077048e7f.PNG)

SimonEit, 2017-09-08:
Yes, i was talking about that dialog @andreasmuelder. Its obviously not a showstopper or something, simply an annoying dialog. Adding a "always save before simulation" checkbox to that dialog would fix it.

closed 2017-09-19 by andreasmuelder