itemisCREATE/statecharts #1277: Red color of active states in a simulation confuses first time users

SimonEit, 2017-03-22:
Multiple usability testusers (who were **first-time-users** of the application)  have mentioned that they associate the red color of active states in a running simulation with things like "error", "problem", "fault", "inactivity" etc. 

Obviously for experienced users this is not a problem. Changing the default color  - for example - from red to green might annoy/confuse this user group.

However i suggest changing the default color and providing the experienced user with the info that he can customize it back the way he is used to via the preferences. This could be done in the release notes for example.

svenjawendler, 2017-05-29:
Here are some suggestions for a new default color for simulation. I prefer orange -> 
1. it is close to red (so the difference is less big) and 
2. it is a part color of the YAKINDU logo and 
3. it is shinny and easily recognized and 
4. people with red-green color blindness recognize this accentuation

[image]
[image]
[image]

BeckmaR, 2017-06-13:
+1 for orange

andreasmuelder, 2017-06-14:
+1 for yellow orange

andreasmuelder, 2017-06-14:
@terfloth please comment

tkutz, 2017-06-14:
+1 for yellow/orange

jdicks, 2017-06-19:
+1 for yellow/orange

svenjawendler, 2017-06-22:
5 votes for orange.

closed 2017-06-26 by svenjawendler