itemisCREATE/statecharts #1282: The hint "Content Assist Available" is too hard to notice

SimonEit, 2017-03-22:
This hint is crucial for a convenient usage of sct but in all of the usability tests not a single user was aware of the lamp-symbol that should inform the user.

To tackle this, there a multiple options:
a) Make the Lamp-Symbol more prominent (i.e. larger in size)
+ Easy to implement
- Might be annoying if you already know about Ctrl+Space

b) Switch to another Symbol which is more noticable but keep the size. For example a bright green exclamation mark
+ Easy to implement
- Might still get overlooked by users

c) Implement it in a way that this hint cannot be overlooked until the user clicks it away
+ This would make sure that the users is aware of this important hint 
- Not entirely show how this can be implemented

svenjawendler, 2017-05-02:
Add the hint as comment to the definition section.

svenjawendler, 2017-06-12:
Suggestion b) should not be realized. It violates the "conformity with user expectations" for eclipse users (ISO 9241-110).

svenjawendler, 2017-06-12:
[image]

PR #1417 (2017-06-12): Fixes #1282 Add hint for Ctrl-Space to comment of definition section
Please merge this file and close #1282
commit ac2fc3a7 (2017-06-12): Fixes #1282 Add hint for Ctrl-Space to comment of definition section (#1417)

* Fixes #1282 Add hint for Ctrl-Space to comment of definition section

* Removed '(CMD)'  because on Mac it is also Ctrl
commit 4304ac40 (2017-06-23): Issue 1277 simulationcolor (#1446)

* Fixes #1282 Add hint for Ctrl-Space to comment of definition section

* Change color to light orange

* Made colors final static
closed 2017-06-12 by andreasmuelder