itemisCREATE/statecharts #1293: Add means for supporting goto on transitions

terfloth, 2017-03-27:
From a forum question:

When the number of transitions is high and the arrows cross each other the diagram becomes difficult to understand. Under this situation it is normal to divide the arrows into two pieces with a label that indicates that both pieces are part of the same transition (connector).

BeckmaR, 2017-03-30:
Something like this? Looks crazy.
[image]

rvilarl, 2017-03-31:
When it gets complex, I believe that it makes sense.
[image]

BeckmaR, 2017-04-01:
Hey rvilarl, thank you for taking the time to draw such a huge example. I can absolutely see that it does make sense for certain transitions. I think it would be nice to have the option "Hide" in the context menu of a transition, and maybe a hidden transition could be shown completely when the user hovers over it... Interesting idea!