itemisCREATE/statecharts #1294: Updating to latest Yakindu breaks active(region.state) references

elgeekomm, 2017-03-28:
After updating Yakindu via Eclipse update, the following code no longer works:

[image]

Previously this Statechart worked correctly, but now any references to a state in another region generate an error. This has broken numerous Statecharts. Is this is a bug, or did the syntax change?

andreasmuelder, 2017-03-28:
@elgeekomm I think you missed the statechart prefix in the reference, this model is working (no errors, can be simulated and generated) with the latest release:

[image]

However, the code completion is broken and the linking when the embedded editor is open seems to be broken. We will fix this ASAP and provide a bugfix release. Thanks for reporting.

andreasmuelder, 2017-03-28:
fixed via PR #1295

elgeekomm, 2017-03-28:
Thanks for the quick response Andreas. I'm afraid this has broken our build. A few questions that may help get us get back online:

1. In which version was the `active(state)` syntax changed? I may have missed this in a changelog or blog.
2. The current documentation on statecharts.org references the old syntax, 6.9.3 `active(state)`. Is this no longer the case?
3. Is the `active(StateIsActive.state)` backwards compatible with older versions of Yakindu?
4. We have a team of 25 developers who use our development environment with varying regularity. We have 18 statecharts in our code repository, each with dozens of states. This change breaks almost all of them. The only resolution that comes to mind is to update every broken statechart with the new syntax and then refactor into any feature branches, and simultaneously have every developer update their Yakindu version. Is there an easier migration path?

Thanks in advance for any advice / suggestions regarding this issue.

andreasmuelder, 2017-03-28:
Too bad with your build, but I am afraid I didn't catch the problem yet.

We didn't change the 'active' syntax. It was always 
active(<statechartname>.<region>.<statename>)
(In my example above the name of the Statechart is 'StateIsActive')

I just checked this with a really old 2.4 release from 2014 and it did not work there, too.
[image]

What I have seen so far is that if you ignore this error message in the embedded editor and close it, the error marker gets away and you can generate code out of it. But this did only work due to a side effect and I doubt that this is the way all your statecharts are created??

So it would be helpful if you can send me a statechart and tell me the exact version you have been using by now.

elgeekomm, 2017-03-28:
I think I understand the problem. It looks like we were taking advantage of namespace resolution, which has recently changed. Up to at least version 2.9.0 (and I think one or two releases after), the following works:

[image]

This diagram generates successfully in 2.9.0 and no problems or errors are reported. On version 2.9.3, an error is shown, and code generation fails:

[image]

> The referenced model(machine) contains errors and could not be generated:
> Couldn't resolve reference to State 'motor.running'.
> Couldn't resolve reference to State 'motor.stopped'.

The resolution is to add the complete scope:

[image]

Model attached that works in 2.9.0 but is broken in 2.9.3.

The consequence is, for example, the guard `active(door.left_open)` becomes `active(machine.main_region.unlocked.unlocked.running.door.left_open)`. This affects dozens of transitions throughout our code.

[diagram.sct.txt](https://github.com/Yakindu/statecharts/files/877008/diagram.sct.txt)

andreasmuelder, 2017-03-28:
We changed this with commit 12b0293ddd2fa677a976353fdb2c490a7994753d as a performance improvement for large models. 
We will restore the old behavior and will also ensure that no errors are reported in the embedded editors. 
I think we can provide a bug fix release until the end of this week. Is this ok for you?

elgeekomm, 2017-03-28:
That will be amazing! Thanks for the quick response. We'll stay tuned to this issue.

andreasmuelder, 2017-04-03:
@elgeekomm I just build an integration release for testing. It would be great if you could install it to see if everything is as you expect. Just download the latest SCT release, click help -> install new software and enter http://updates.yakindu.org/sct/integration as the work with URL.

elgeekomm, 2017-04-03:
Beautiful. Installed the 2017-04-03 integration release and everything works great.

andreasmuelder, 2017-04-06:
Unfortunately this commit did not pass our QA due to some performance drawbacks with large models.
I will release a bugfix release with the old behavior prior 2.9.0 today to ensure that your models are working and create a new fix for the error in the embedded editor for the upcoming release scheduled for 28th of April.

elgeekomm, 2017-04-06:
Thanks Adreas, this will work for us. Are you pushing the bug fix to the main install path (i.e. just a standard Eclipse update)?

andreasmuelder, 2017-04-07:
Yes it should now be available via eclipse update

elgeekomm, 2017-04-07:
I installed the 2017-04-06 eclipse update and the build works, with one exception. I noticed with both the 2017-04-03 integration release, and with the 2017-04-06 release, the attached model is broken. It's derived from a model that worked in previous releases. I have to comment out a line for it to work now.

[image]

[default.sct.txt](https://github.com/Yakindu/statecharts/files/906255/default.sct.txt)

andreasmuelder, 2017-04-10:
@elgeekomm It would be great if you could create a new issue for that. We can fix this for the upcoming planned release in 4 weeks.

elgeekomm, 2017-04-10:
@andreasmuelder done! https://github.com/Yakindu/statecharts/issues/1310

BeckmaR, 2017-06-20:
@andreasmuelder, @elgeekomm - please clarify the current state of this issue. What needs to be done?

elgeekomm, 2017-06-20:
I believe this issue has been resolved. The latest version of the tools is working well for us.

andreasmuelder, 2017-06-27:
It is not completely fixed yet - it still shows an error in the embedded editor. But at least it is not a high priority issue anymore.

andreasmuelder, 2017-09-20:
Latest version does support implicit imports for parent region names and navigation to referenced state with CTRL + Click

closed 2017-09-20 by andreasmuelder