itemisCREATE/statecharts #1310: Event name resolves in parent state but shows undefined in composite state

elgeekomm, 2017-04-10:
From the 2017-04-06 eclipse update, the attached model is broken. It's derived from a model that worked in previous releases. I have to comment out a line for it to work now.

[image]

Note that the transition in the parent state references "request_run" without error, however a reference to the same event name in the composite state produces an error.

Model attached.

[default.sct.txt](https://github.com/Yakindu/statecharts/files/910874/default.sct.txt)

BeckmaR, 2017-04-11:
Hey @elgeekomm,

actually, the model never should have worked because the syntax is wrong.
See the [documentation about reactions](https://www.itemis.com/en/yakindu/state-machine/documentation/user-guide/#reactions). 
Basically it's `trigger [guard] / action` - you use `request_run` as a guard, but it's not usable as such because it's an event. You need to use it instead of `oncycle`, so the reaction should look like this:
`request_run / print("Already ran")`.

Please report back if this works out.

Regards,
Rene

elgeekomm, 2017-04-11:
You're correct, I was using the incorrect syntax. Thanks for pointing that
out. Strange, this model worked for about a year and only recently broke!
We can close out this issue.

------------------------------

[image: Momentum Machines] <http://momentummachines.com>

Jeff C. Jensen
Momentum Machines

[image: LinkedIn] <https://linkedin.com/in/elgeeko> [image: Twitter]
<https://twitter.com/elgeeko> [image: Zoom]
<https://momentummachines.zoom.us/j/3050957577>

elgeekomm, 2017-04-11:
Correction, I do think there is an issue here. Here's a refinement of the model that reports an unusual error.

[image]

fixed by removing "oncycle" from the request_run guard.

[image]

[default.sct.txt](https://github.com/Yakindu/statecharts/files/914331/default.sct.txt)

Notice that the events request_run, x and y are all in the same scope, yet "oncycle" produces an error only in the case of request_run. I can fix this in my model by removing oncycle for the request_run action, as you suggested.

I found I need to use oncycle if I want an action to occur if one or more events occur, is this the recommended syntax?

BeckmaR, 2017-04-25:
You don't need to use oncycle, but you would need the square brackets for multiple events, that is indeed true, because you do not use them as triggers but the implicit boolean that flags if they are raised or not. So request_run should be available inside of the square brackets as well as in place of the trigger, but you do not need to specify oncycle or always if you have a guard in place. You just need at least one of both.

terfloth, 2017-07-20:
Seems like the scoping within guard expressions has a defect.

andreasmuelder, 2017-08-01:
Could not figure out how this regression was introduced. Fixed with PR #1573

PR #1573 (2017-08-01): Make Events visible in context of a Local Reaction
Fixes #1310
closed 2017-08-01 by andreasmuelder