itemisCREATE/statecharts #1318: isActive() not working on a non initialised machine

wixiw, 2017-04-23:
Hi,

The isActive() generated method is like this 
```
typedef enum
{
	main_region_Waiting_table_installation,
	FSM_Lifecycle_last_state
} FSM_LifecycleStates;

sc_boolean FSM_Lifecycle::isActive()
{
	return stateConfVector[0] != FSM_Lifecycle_last_state;
}
```

The "[machine]States" enum generated with each machine doesn't have any default value. Hence, the constructor will in most case take the first state as the default value.

In my case  : stateConfVector[0] = 0 = main_region_Waiting_table_installation

A call to the init() method will prevent the problem to happen as it initialisation the state to FSM_Lifecycle_last_state.

If I can understand that it's working if the user following the API contract, I would expect a more robust behavior from Yakindu in case init() is never called, as in complex project it's quite difficult to find such a bug.

I would suggest to add a default invalid value with label 0 to state enums and to use it instead of FSM_Lifecycle_last_state which is not done for that.

```
typedef enum
{
        FSM_Lifecycle_default,
	main_region_Waiting_table_installation,
	FSM_Lifecycle_last_state
} FSM_LifecycleStates;

sc_boolean FSM_Lifecycle::isActive()
{
	return stateConfVector[0] != FSM_Lifecycle_default;
}
```

BeckmaR, 2017-05-31:
Closed with #1391

closed 2017-05-31 by BeckmaR

paphko, 2017-08-15:
I just had a discussion with @DefCon-CC who wonders why 'last_state' is now generated as **first** enum literal [here](https://github.com/Yakindu/statecharts/commit/96de1fa47cd0f079ae633edd6d8440179bd55858#diff-0dc78b197a95e061606395318c4f4938R126). That looks weird to us in the generated code. @BeckmaR or @andreasmuelder, could you comment on that?

terfloth, 2017-08-15:
@paphko @DefCon-CC : indeed the fact that 'last_state' is the first one is confusing. Th change was introduced while fixing this issue. From a functional point of view the change makes sens. The 'last_state' was never used as some last index in loops but only in the sense of 'no_state'. And 'no_state' should better have value '0' for a consistent initialization etc.

So basically it was missed to rename the value to 'no_state'. That should make the intent clear.