itemisCREATE/statecharts #1322: Flakey update of error decorators

jdicks, 2017-04-24:
The time consumed to update the decorators of the editor is unconstant.

How to reproduce: 
* use a big model file (attached)
* change a value within an embeded editor

The time needed until the job 'validate model' job is finished differs from <1sec to >20sec.

jdicks, 2017-04-24:
test model:
[big_file.zip](https://github.com/Yakindu/statecharts/files/952129/big_file.zip)

jdicks, 2017-04-25:
found this stack trace:
```
!ENTRY org.eclipse.gmf.runtime.diagram.ui 4 4 2017-04-25 11:41:37.598
!MESSAGE execute
!STACK 0
org.eclipse.core.commands.ExecutionException: While executing the operation, an exception occurred
	at org.eclipse.core.commands.operations.DefaultOperationHistory.execute(DefaultOperationHistory.java:496)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramCommandStack.execute(DiagramCommandStack.java:206)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramCommandStack.execute(DiagramCommandStack.java:169)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramCommandStack.execute(DiagramCommandStack.java:156)
	at org.yakindu.base.xtext.utils.gmf.directedit.DirectEditManagerEx.commit(DirectEditManagerEx.java:146)
	at org.yakindu.base.xtext.utils.gmf.directedit.XtextDirectEditManager.commit(XtextDirectEditManager.java:373)
	at org.yakindu.base.xtext.utils.gmf.directedit.DirectEditManagerEx$3.applyEditorValue(DirectEditManagerEx.java:293)
	at org.eclipse.jface.viewers.CellEditor$1.run(CellEditor.java:336)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.ui.internal.JFaceUtil$1.run(JFaceUtil.java:50)
	at org.eclipse.jface.util.SafeRunnable.run(SafeRunnable.java:173)
	at org.eclipse.jface.viewers.CellEditor.fireApplyEditorValue(CellEditor.java:333)
	at org.eclipse.jface.viewers.CellEditor.focusLost(CellEditor.java:722)
	at org.yakindu.base.xtext.utils.jface.viewers.XtextStyledTextCellEditor.focusLost(XtextStyledTextCellEditor.java:212)
	at org.yakindu.base.utils.jface.viewers.StyledTextCellEditor.access$3(StyledTextCellEditor.java:1)
	at org.yakindu.base.utils.jface.viewers.StyledTextCellEditor$5.focusLost(StyledTextCellEditor.java:157)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:144)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4418)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1079)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1103)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1084)
	at org.eclipse.swt.widgets.Control.sendFocusEvent(Control.java:2934)
	at org.eclipse.swt.widgets.Widget.wmKillFocus(Widget.java:1961)
	at org.eclipse.swt.widgets.Control.WM_KILLFOCUS(Control.java:5180)
	at org.eclipse.swt.widgets.Canvas.WM_KILLFOCUS(Canvas.java:469)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4833)
	at org.eclipse.swt.widgets.Canvas.windowProc(Canvas.java:359)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5123)
	at org.eclipse.swt.internal.win32.OS.SetFocus(Native Method)
	at org.eclipse.swt.widgets.Control.forceFocus(Control.java:1114)
	at org.eclipse.swt.widgets.Control.setFocus(Control.java:3447)
	at org.eclipse.swt.widgets.Composite.setFocus(Composite.java:1124)
	at org.yakindu.base.utils.jface.viewers.StyledTextCellEditor.doSetFocus(StyledTextCellEditor.java:193)
	at org.eclipse.jface.viewers.CellEditor.setFocus(CellEditor.java:839)
	at org.yakindu.base.xtext.utils.gmf.directedit.DirectEditManagerEx.show(DirectEditManagerEx.java:406)
	at org.yakindu.base.xtext.utils.gmf.directedit.XtextDirectEditManager.show(XtextDirectEditManager.java:518)
	at org.yakindu.base.xtext.utils.gmf.directedit.XtextLabelEditPart$2.run(XtextLabelEditPart.java:133)
	at org.eclipse.emf.transaction.impl.TransactionalEditingDomainImpl.runExclusive(TransactionalEditingDomainImpl.java:328)
	at org.yakindu.base.xtext.utils.gmf.directedit.XtextLabelEditPart.performDirectEditRequest(XtextLabelEditPart.java:118)
	at org.yakindu.base.xtext.utils.gmf.directedit.XtextLabelEditPart$1.handleDoubleClick(XtextLabelEditPart.java:107)
	at org.yakindu.base.xtext.utils.gmf.directedit.DoubleClickDirectEditDragTracker.handleDoubleClick(DoubleClickDirectEditDragTracker.java:41)
	at org.eclipse.gef.tools.AbstractTool.mouseDoubleClick(AbstractTool.java:1069)
	at org.eclipse.gef.tools.SelectionTool.mouseDoubleClick(SelectionTool.java:525)
	at org.eclipse.gef.tools.SelectionTool.mouseDoubleClick(SelectionTool.java:527)
	at org.eclipse.gef.EditDomain.mouseDoubleClick(EditDomain.java:231)
	at org.eclipse.gef.ui.parts.DomainEventDispatcher.dispatchMouseDoubleClicked(DomainEventDispatcher.java:291)
	at org.eclipse.draw2d.LightweightSystem$EventHandler.mouseDoubleClick(LightweightSystem.java:518)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:197)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4418)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1079)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4236)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3824)
	at org.eclipse.ui.internal.dialogs.EventLoopProgressMonitor.runEventLoop(EventLoopProgressMonitor.java:125)
	at org.eclipse.ui.internal.dialogs.EventLoopProgressMonitor.isCanceled(EventLoopProgressMonitor.java:99)
	at org.eclipse.core.internal.jobs.ThreadJob.isCanceled(ThreadJob.java:144)
	at org.eclipse.core.internal.jobs.ThreadJob.waitForRun(ThreadJob.java:233)
	at org.eclipse.core.internal.jobs.ThreadJob.joinRun(ThreadJob.java:197)
	at org.eclipse.core.internal.jobs.ImplicitJobs.begin(ImplicitJobs.java:92)
	at org.eclipse.core.internal.jobs.JobManager.beginRule(JobManager.java:307)
	at org.eclipse.emf.transaction.util.Lock.uiSafeAcquire(Lock.java:359)
[... cut ...]
	at org.eclipse.emf.transaction.impl.TransactionImpl.start(TransactionImpl.java:204)
	at org.eclipse.emf.transaction.impl.TransactionalEditingDomainImpl.startTransaction(TransactionalEditingDomainImpl.java:424)
	at org.eclipse.emf.workspace.AbstractEMFOperation.createTransaction(AbstractEMFOperation.java:617)
	at org.eclipse.emf.workspace.AbstractEMFOperation.execute(AbstractEMFOperation.java:147)
	at org.eclipse.core.commands.operations.DefaultOperationHistory.execute(DefaultOperationHistory.java:488)
	... 101 more
```

BeckmaR, 2017-04-25:
Could not recreate a deadlock. Validating times varied, but don't they always do in Eclipse? :laughing:

jdicks, 2017-04-25:
i could not reproduce unstable validation times anymore - closed.

closed 2017-04-25 by jdicks