itemisCREATE/statecharts #1332: StateMachine interface depends on the content

wixiw, 2017-05-01:
Depending on the statemachine content (if you have a timed transition or not), the State machine has a different interface. It's causing issues when machines are integrated in a decorated environment.

It's annoying that when you delete your last transition (maybe be done by "non-SW" people in the IDE), it causes the project to fail its build.

Moreover as the "timer" pointer is never checked, it's easy to forgot to register it if you just add the first time transition. 

It would suggest to add a mean to always generate state machine against the timed interface (even if no timed event are set), AND please do check pointers before use. An smart way of doing it would be during the init() or enter().

BeckmaR, 2017-05-02:
+1 for generating the event even if timed transitions are not used. I already fell for that trap and it is annoying to adapt the warpping sourcecode when the statechart changes. That's completely contrary to our idea of simple integration.

terfloth, 2017-07-20:
The observed behavior was implemented with full intent ;-). The main guideline is to genrate only those things that are required and nothing more.

An additional option could may sense. Nevertheless we have to think about that in order to avoid the bloat of sgen configuration options. May be we could define a @Timed annotation alternatively  -what leads to bloat of annotations... :-) 

We will consider this.