itemisCREATE/statecharts #1345: [C++] A maj constant has a badly named accessor

wixiw, 2017-05-04:
`var readonly SUCCESS : integer = 1`
is generated as :
`sc_integer get_sUCCESS();`

whereas I would expect 
`sc_integer get_SUCCESS();`

BeckmaR, 2017-05-09:
The generated code is quite ugly indeed, but @terfloth needs to decide if we should change that - we might break client code that uses the ugly function already.

terfloth, 2017-05-09:
I totally agree with the user expectation! The naming should be as consistent as possible with the naming in the statechart. 

BUT we have to consider:
1.  Backwards compatibility - we cannot break existing generated API by simply changing the naming style. 
2. Introducing yet another configuration options bloats generator configurations.

Introducing a new version of the code generator and retaining the current one could be an option. In this context we could also consider other issues and feature requests.

terfloth, 2017-05-10:
@wixiw : just to mention it - as a user you could of course also implement a custom code generator derived from our standard C++ code generator. This custom generator could implement any naming convention required.

wixiw, 2017-05-10:
Great ! Hence, it would be easy for you to break the client code API as they will be able to implement their own generator to fit to the legacy solution ;-)

andreasmuelder, 2018-01-02:
We could add a second getter with the right name, delegate from the old getter to the new getter and deprecate the old one. Then we can remove it in the next version.

Implementing a new code generator just for the "right" naming convention seems to be too much overhead.

BeckmaR, 2018-01-08:
There is one thing we'd have to pay attention to - if the user has, for whatever reason, a variable named `SUCCESS` and one named `sUCCESS`, or maybe just `x` and `X`, which is far more likely, this would break everything. So we'd at least have to check for this oddity.

BeckmaR, 2018-01-26:
Also, there is no really portable way to mark functions as deprecated in C++. There are some macros floating across the web but none are guaranteed to work in every compiler.

wixiw, 2018-01-26:
Often, the concept a deprecation is just a matter of documentation. 
If you want to go further you can have some custom macros "#define DEPRECATED_API" and surround any deprecated function with this. Then the user can choose to define or not the deprecation flaga t compilation. It's a quite "C" way of doing, not very pleasant, but it works. 

BeckmaR made a very important point stating that two variable differing only with case sensivity could provoke a clash at generation time. or provoke hard-to-find bugs for the user.

BeckmaR, 2018-01-26:
I've thought about the thing with variables that only differ in case, and basically it's a problem already - if someone names a variable x and one X, both would be made `toFirstLower`, which won't work. So, in a first step I'm currently implementing a warning that comes up when variables / all named things only differ in case.

BeckmaR, 2018-01-26:
[image]

That's what it looks like, I think that might work.

BeckmaR, 2018-04-18:
In what cases should we have the second getter? For constants, for variables with all uppercase, ...?

BeckmaR, 2019-07-15:
Closing b/c outdated

closed 2019-07-15 by BeckmaR