itemisCREATE/statecharts #1346: [C++] Access to generated constant is unpractical

wixiw, 2017-05-04:
When declaring a constant like :
`var readonly NONE : integer = 0`
The associated getter is :
`sc_integer get_nONE();`

I would expect to either have :
- a public const member (prefered)
- a public const accessor
So that it's possible to use it in switch cases.

terfloth, 2017-05-06:
The `readonly` keyword restricts accessibility of the client code. It does not imply read only access for the state machines implementation.

We should introduce const to the getter signature.

wixiw, 2017-05-07:
When I write 
`internal :
   var readonly ....`
I expect the variable to be in read only access in the state machine implementation. It may be a langage extrapolation, but I think we need in anyway a mean to create constants for the state machine.

Behind all that, my use case is that I'd like to share enums between the state machine and the client code. My solution was to use integer in operations signatures, and to create readonly vars that I would use in client code instead of the error.

The const keyword is important as it helps the compiler to better optimize thing.

wixiw, 2017-05-07:
I just discovered the const keyword from documentation that allows a better fit to my need, but as you noticed, the const is missing on the getter 
`const sc_integer FSM_Lifecycle::get_mODE_NONE()
{
	return FSM_Lifecycle::DefaultSCI::MODE_NONE;
}`

as I'd like to have 
`
const sc_integer FSM_Lifecycle::get_mODE_NONE() const
{
	return FSM_Lifecycle::DefaultSCI::MODE_NONE;
}
`

BeckmaR, 2017-05-09:
Was fixed with 02beebe, is not released yet though.

closed 2017-05-09 by BeckmaR