itemisCREATE/statecharts #1348: (Eclipse sct editor) Impossible to save an sct when the focus is in the interface widget

wixiw, 2017-05-04:
When the interface widget is focused in the state machine editor view, it's impossible to save with Ctrl+S. 
I need to clic out of the interface widget, then stroke Ctrl+S.

As in the same time, the eclipse tab is not tagged as changed (with a little (*) next to the fsm name), it's very easy to forgot to save the machine when switching to another tab.

andreasmuelder, 2017-05-05:
Thanks for reporting.
This is a duplicate to https://github.com/Yakindu/statecharts/issues/276
Feel free to vote on the original issue

closed 2017-05-05 by andreasmuelder

wixiw, 2017-05-05:
Ok, sorry for duplication, I didn't searched so far in the history

andreasmuelder, 2017-05-05:
We are very happy about any report, we don't expect you to read the whole backlog before reporting  :smile: